zephyrproject-rtos/zephyr #32959: [Coverity CID :219480] Out-of-bounds read in lib/os/cbprintf_nano.c

zephyrbot, 2021-03-05:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/lib/os/cbprintf_nano.c

Category: Memory - illegal accesses
Function: `cbvprintf`
Component: Other
CID: [219480](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=219480)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/lib/os/cbprintf_nano.c

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

pabigot, 2021-03-05:
This is an out-of-bounds read at line 208 where `convert_value` accesses `buf[21]` in a 21-byte buffer introduced in fb73ac392c27328cddae5622a3482d35d09d99ad.

On short inspection it seems to be a false positive, but it's not the sort of mistake Coverity should make.

npitre, 2021-03-05:
This is clearly a false positive. Coverity probably isn't smart enough 
to go within the called function and see that the buffer is used 
downwards with a pre-decremented index.

npitre, 2021-03-12:
The coverity link is giving me a 401.
As I said, this is a false positive.
Closing.

closed 2021-03-12 by npitre