zephyrproject-rtos/zephyr #32963: USB device is not supported by qemu_x86 platform

Zhaoningx, 2021-03-05:
Describe the bug
There is a question or bug about qemu_x86 platform. I need to use qemu_x86 to verify tracing's usb backends. But I find it doesn't work, and it showed me 'usb' not supported in zephyr doc. See link: https://docs.zephyrproject.org/latest/boards/x86/qemu_x86/doc/index.html 
And qemu_x86 board configure as below:
```
identifier: qemu_x86
name: QEMU Emulation for X86
type: qemu
simulation: qemu
arch: x86
toolchain:
  - zephyr
  - xtools
  - llvm
supported:
  - pci
  - nvs
  - netif:serial-net
  - ieee802154
  - eeprom
testing:
  default: true

```

So who knows how to configure USB device on qemu_x86 platform, or is USB device supported by qemu_x86 or not?

jfischer-no, 2021-03-08:
There is no USB device support for any qemu platform.

closed 2021-03-08 by jfischer-no

Zhaoningx, 2021-03-09:
Hi @jfischer-no ,  it is confusing, you mean usb device cannot be supported by qemu_x86 whatever you do? Or you can add some configures to enable usb device?

Zhaoningx, 2021-03-11:
Hi @jfischer-no , Could you comment my question above? Thanks

jfischer-no, 2021-03-12:
> Hi @jfischer-no , it is confusing, you mean usb device cannot be supported by qemu_x86 whatever you do? Or you can add some configures to enable usb device?

I do not know if qemu is able to emulate USB device controller or whatever, that also does not matter since "There is no USB device support for any qemu platform" in current code base. No drivers. There might be another way to make USB accessible for qemu but I do not want to promise it now and I have to test it first myself.