zephyrproject-rtos/zephyr #32966: Run some testcases timeout failed by using twister

Zhaoningx, 2021-03-05:
Describe the bug:
I find some testcases timeout failed after twister changed. More detail as below:

**frdm_k64f platform:**
Just select a testcase to demo this failure.
command: twister -p frdm_k64f --device-testing --device-serial /dev/ttyACM0 -T tests/kernel/stack/ -j 1 -vv 
and it showed me timeout failed, and device.log shows:
```
WARNING: runners.pyocd: hex file (None) does not exist; falling back on .bin (/home/zn/zephyrproject/zephyr/twister-out/frdm_k64f/tests/kernel/stack/stack/kernel.stack.usage/zephyr/zephyr.bin). Consider enabling CONFIG_BUILD_OUTPUT_HEX.
0005001:INFO:loader:Erased 77824 bytes (19 sectors), programmed 77824 bytes (19 pages), skipped 4096 bytes (1 page) at 19.53 kB/s
```
**And I reset to commit c5caa9b916d1 ,  It is good to work.**

**sam_e70b_xplained platform:**
command: twister -p sam_e70b_xplained --device-testing --device-serial /dev/ttyACM1 -T tests/kernel/stack/stack/  -j 1 -vv
and it showed me timeout failed, device.log shows:
```
Open On-Chip Debugger 0.10.0+dev-01508-gf79c90268-dirty (2021-01-24-19:03)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : flash bank command
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: FW Version = 03.25.01B6
Info : CMSIS-DAP: Serial# = ATML2637010000000000
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 1 TDO = 1 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 1800 kHz
Info : SWD DPIDR 0x0bd11477
Info : atsame70q21.cpu: hardware has 8 breakpoints, 4 watchpoints
Info : starting gdb server for atsame70q21.cpu on 3333
Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* atsame70q21.cpu    cortex_m   little atsame70q21.cpu    running

target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00403638 msp: 0x20402c00
Info : device id = 0xa1020e01
Info : erasing lock regions 0-4...
Info : erasing lock region 0
Info : erasing lock region 1
Info : erasing lock region 2
Info : erasing lock region 3
Info : erasing lock region 4
auto erase enabled
wrote 81920 bytes from file /home/xuhao/work/zephyrproject/zephyr/twister-out/sam_e70b_xplained/tests/kernel/stack/stack/kernel.stack.usage/zephyr/zephyr.hex in 1.864619s (42.904 KiB/s)

target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00402ec0 msp: 0x20402400
verified 70972 bytes in 1.719761s (40.301 KiB/s)

shutdown command invoked
```
**And I reset to commit c5caa9b916d1 ,  It is good to work.**

**mimxrt1050_evk platform:**
command: twister -p mimxrt1050_evk --device-testing --device-serial /dev/ttyACM0 -T tests/kernel/stack/stack/  -j 1 -vv
and it showed me build failed, and there is nothing wrong in build.log file.
**And I reset to commit c5caa9b916d1 ,  It is good to work.**

And reel_board platform is also not work, more detail refer above.

Environment
OS: Fedora33
Toolchain:zephyr-sdk-0.12.2
Commit ID: 4118ed1d4d6fe50f4

nashif, 2021-03-09:
just bisected to 

```
commit f1a27a8189e338f2c2d59c146002952bd9537feb
Author: Ioannis Glaropoulos <Ioannis.Glaropoulos@nordicsemi.no>
Date:   Mon Mar 1 11:25:49 2021 +0100

    arm: cortex_m: assert if DebugMonitor exc is enabled in debug mode

    Assert if the null pointer de-referencing detection (via DWT) is
    enabled when the processor is in debug mode, because the debug
    monitor exception can not be triggered in debug mode (i.e. the
    behavior is unpredictable). Add a note in the Kconfig definition
    of the null-pointer detection implementation via DWT, stressing
    that the solution requires the core be in normal mode.

    Signed-off-by: Ioannis Glaropoulos <Ioannis.Glaropoulos@nordicsemi.no>

 arch/arm/core/aarch32/cortex_m/Kconfig  | 1 +
 arch/arm/include/aarch32/cortex_m/dwt.h | 8 ++++++++
 2 files changed, 9 insertions(+)
```

LeiW000, 2021-03-10:
@hakehuang @ioannisg , could you take look at it? It blocks the testing with twister on almost all ARM boards.

ioannisg, 2021-03-10:
@LeiW000 isn't this fixed now in master?

Zhaoningx, 2021-03-10:
Hi @ioannisg @LeiW000 , just now, I had a try to run testcases on these platforms manually, it seems frdm_k64f , sam_e70b_xplained and mimxrt1050_evk are good to work in master, but reel_board is still not working, it showed my timeout failure, and the device.log info:

```
[ztest@localhost zephyr]$ vi /home/ztest/zephyrproject/zephyr/twister-out/reel_board/tests/kernel/stack/stack/kernel.stack.usage/device.log
0004915:INFO:loader:Erased 81920 bytes (20 sectors), programmed 81920 bytes (20 pages), skipped 4096 bytes (1 page) at 19.21 kB/s

```

hakehuang, 2021-03-10:
@LeiW000  this happens on all NXP platforms

LeiW000, 2021-03-11:
It looks that the issue is fixed partially. We still see it on reel_board.

ioannisg, 2021-03-12:
> It looks that the issue is fixed partially. We still see it on reel_board.

Confirmed. I'll share a fix to this for us to test out.

ioannisg, 2021-03-16:
@LeiW000 I submitted a PR that addresses the issue with the reel_board (essentially, a pyocd issue). Could you please check if you can run CI on reel_board using #33386 ?

PR #33386 (2021-03-17): Reel board custom config pyocd
Fixes #32966 (the fix is for the reel_board). 
The PR
- adds support in pyocd runner to attach a board-specific configuration file
- use a board-specific pyocd.yaml file for reel_board, to pass configuration for the flash command
closed 2021-03-17 by nashif