zephyrproject-rtos/zephyr #32975: Where should a few include/ headers live

galak, 2021-03-05:
There are a few headers in include/ that seem like they shouldn't be at the top level:

`include/ec_host_cmd.h`
`include/emul.h`
`include/ptp_clock.h`

Should these files still live here are be moved under some sub-dir.

galak, 2021-03-05:
@jukkar should `ptp_clock.h` be `include/net/ptp_clock.h`?

galak, 2021-03-05:
@sjg20 should `emul.h` be `include/drivers/emul.h`?

galak, 2021-03-05:
@sjg20 should `ec_host_cmd.h` be `include/mgmt/ec_host_cmd.h`?

jukkar, 2021-03-05:
> should `ptp_clock.h` be `include/net/ptp_clock.h`?

I would put that to `include/drivers/` directory

sjg20, 2021-03-05:
> @sjg20 should `emul.h` be `include/drivers/emul.h`?

Yes that seems fine

sjg20, 2021-03-05:
> @sjg20 should `ec_host_cmd.h` be `include/mgmt/ec_host_cmd.h`?

Yes OK

closed 2021-03-24 by galak