zephyrproject-rtos/zephyr #32984: West: openocd runner: Don't let debug mode on by default

erwango, 2021-03-05:
**Describe the bug**
When flashing using openocd runner, debug mode is set by default. It is the released after a cold reboot.
This conflicts with the use of https://github.com/zephyrproject-rtos/zephyr/pull/31481 (Null pointer exception detection),
and prevent its use by default in tests.
For this reason, Null pointer exception detection was disabled by default on all STM32 platforms, but it would be great to be able to re-instantiate it.

Note: This potentially affect other runners as well (pyocd, for instance).

**To Reproduce**
Steps to reproduce the behavior:
1. Revert https://github.com/zephyrproject-rtos/zephyr/pull/32983
2. Run any zephyr test on a STM32 target using openocd runner for flashing
4. See error

**Expected behavior**
Don't prevent use of Null pointer exception detection when using openocd runner

**Impact**
Prevent use of Null pointer exception detection when using openocd runner in zephyr tests

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

ioannisg, 2021-03-05:
Thanks for the ticket @erwango 
To the best of my knowledge, pyocd, jlink, or nrfjprog are not affected. I've tried with several boards/runners and this seems to be the case.

ioannisg, 2021-03-05:
Guess the main issue here is that we do not have a consistent west flash, with respect to the state the core is left at, after west flash is done.

ioannisg, 2021-03-05:
@mbolivar-nordic can you take a look here (or suggest alternative maintainer)?

nashif, 2021-03-09:
looks like pyocd is also affected

see https://github.com/zephyrproject-rtos/zephyr/issues/32966

mbolivar-nordic, 2021-03-09:
> @mbolivar-nordic can you take a look here (or suggest alternative maintainer)?

Sorry @ioannisg, I don't know enough about the underlying debug IPs to really help here. I see you already posted a PR for openocd, so it seems like you're on your own here :(.

closed 2021-08-31 by github-actions[bot]