zephyrproject-rtos/zephyr #32994: Question: Possible simplification in mutex.h?

GunZi200, 2021-03-05:
Hello, I have a question regarding the code at `include/sys/mutex.h`: https://github.com/zephyrproject-rtos/zephyr/blob/1337f7fe245b8dbdf0d70699f74d35589ed1ec61/include/sys/mutex.h#L157

Can we simplify this.... (see below)
```c
static inline int sys_mutex_unlock(struct sys_mutex *mutex)
{
	if (mutex->kernel_mutex.lock_count == 0) {
		return -EINVAL;
	}

	if (mutex->kernel_mutex.owner != _current) {
		return -EPERM;
	}

	k_mutex_unlock(&mutex->kernel_mutex);
	return 0;
}
```

....To this?

```c
static inline int sys_mutex_unlock(struct sys_mutex *mutex)
{
	return k_mutex_unlock(&mutex->kernel_mutex);
}
```

According to the documentation `k_mutex_unlock()` already returns `EINVAL` and `EPERM` so these additional `if()` statements look redundant to me.

`k_mutex_unlock()` docs: https://docs.zephyrproject.org/latest/reference/kernel/synchronization/mutexes.html#c.k_mutex_unlock

----

Another side point:

`sys_mutex_unlock()` docs: https://docs.zephyrproject.org/latest/reference/kernel/synchronization/mutexes.html#c.sys_mutex_unlock

`sys_mutex_unlock()` returns `-EACCESS` (typo in docs, should be `-EACCES` no?), when using `k_mutex_unlock()` this is not true. Does this apply to `z_sys_mutex_kernel_unlock()`? I could not see it's source so am not sure.

GunZi200, 2021-03-05:
I am now also thinking that in `lib/os/mutex.c`: https://github.com/zephyrproject-rtos/zephyr/blob/074dbb9982ad2e4158b92c9aff174969b72402c3/lib/os/mutex.c#L55

Can be simplified from:

```c
int z_impl_z_sys_mutex_kernel_unlock(struct sys_mutex *mutex)
{
	struct k_mutex *kernel_mutex = get_k_mutex(mutex);

	if (kernel_mutex == NULL || kernel_mutex->lock_count == 0) {
		return -EINVAL;
	}

	if (kernel_mutex->owner != _current) {
		return -EPERM;
	}

	k_mutex_unlock(kernel_mutex);
	return 0;
}
```

To this:

```c
int z_impl_z_sys_mutex_kernel_unlock(struct sys_mutex *mutex)
{
	struct k_mutex *kernel_mutex = get_k_mutex(mutex);

	if (kernel_mutex == NULL || kernel_mutex->lock_count == 0) {
		return -EINVAL;
	}

	return k_mutex_unlock(kernel_mutex);
}
```
Here I am assuming the file at `kernel/mutex.c` defines the `k_mutex_unlock()` call https://github.com/zephyrproject-rtos/zephyr/blob/074dbb9982ad2e4158b92c9aff174969b72402c3/kernel/mutex.c#L212

nashif, 2021-03-05:
> 
> sys_mutex_unlock
> ```

yes, this can be simplified. The error checking in kernel/mutex.c came after, so now we have it duplicated. Please submit a PR :-)

PR #33002 (2021-03-10): Simplify code where k_mutex_unlock() is used and small documentation fixes
Fix #32994

* docs: Fix typo in mutex docs 
  * `-EACCESS` should be `-EACCES`
* docs: Add missing retval for `sys_mutex_unlock() `
  *  Should return 0 if mutex is unlocked, like `k_mutex_unlock()`
* include: sys: Simplify `sys_mutex_unlock()`
  * Remove redundant `if()` statements that are already included with `k_mutex_unlock()`
* lib: os: Simplify `z_impl_z_sys_mutex_kernel_unlock()`
  * Remove one redundant `if()` statement already included when `k_mutex_unlock()` is used