zephyrproject-rtos/zephyr #33009: kernel: k_heap failures on small heaps

JordanYates, 2021-03-06:
**Describe the bug**

`k_heap_alloc` fails when attempting to allocate memory from small heaps.
The following code will fail:
```
K_HEAP_DEFINE(test_heap, 40);

void main(void) {
    void * p = k_heap_alloc(&test_heap, 8, K_NO_WAIT);
    printk("%p\n", p);
}
```
On a Cortex M4F board, this will just return NULL, but on `native_posix_64` a `K_ERR_KERNEL_PANIC` is triggered.
Based on the documentation (https://docs.zephyrproject.org/latest/reference/kernel/memory/heap.html), it seems clear that there is simply not enough space at the start of the buffer to hold the bucket list for allocations.

The bug is not that the k_heap cannot support tiny heaps, but that fatal errors can occur.
It also not obvious from the documentation how large the heap needs to be before it will start working.
64 bytes is sufficient to allocate 8 bytes on a Cortex M4F , but not `native_posix_64`.

**Expected behavior**

Allocating from small heap should just return NULL, not hard fault.
Ideally some form of validation would be done on the values provided to `K_HEAP_DEFINE` to ensure that it is large enough to return a valid 8 byte (smallest allocation unit) memory chunk at least once.

Something similar to:
```
#define K_HEAP_DEFINE(name, bytes)				\
	BUILD_ASSERT((bytes) >= (MINIMUM_BUCKET_SIZE + 8), "Heap too small to allocate memory from");\
	char __aligned(sizeof(void *)) kheap_##name[bytes];	\
	Z_STRUCT_SECTION_ITERABLE(k_heap, name) = {		\
		.heap = {					\
			.init_mem = kheap_##name,		\
			.init_bytes = (bytes),			\
		 },						\
	}
```

**Environment (please complete the following information):**
 - Zephyr v2.5

yerabolu, 2021-03-17:
just a quick update @andyross: tried running the test with a small heap size of 40 and tests passed on both Cortex_M3 and native_posix_64.

JordanYates, 2021-03-17:
On the latest master with the following diff, I can still reproduce the fatal errors
```
jordan@DOTA:/mnt/c/Users/Jordan/code/zephyr/zephyr$ git diff HEAD
diff --git a/tests/kernel/mem_heap/k_heap_api/src/test_kheap_api.c b/tests/kernel/mem_heap/k_heap_api/src/test_kheap_api.c
index dc7d0c2ead..d8a93cab9b 100644
--- a/tests/kernel/mem_heap/k_heap_api/src/test_kheap_api.c
+++ b/tests/kernel/mem_heap/k_heap_api/src/test_kheap_api.c
@@ -13,6 +13,7 @@ K_THREAD_STACK_DEFINE(tstack, STACK_SIZE);
 struct k_thread tdata;

 K_HEAP_DEFINE(k_heap_test, HEAP_SIZE);
+K_HEAP_DEFINE(k_heap_test_small, 40);

 #define ALLOC_SIZE_1 1024
 #define ALLOC_SIZE_2 1536
@@ -61,6 +62,9 @@ void test_k_heap_alloc(void)
                p[i] = '0';
        }
        k_heap_free(&k_heap_test, p);
+
+       p = (char *)k_heap_alloc(&k_heap_test_small, 8, timeout);
+       zassert_not_null(p, "k_heap_alloc operation failed");
 }
```

PR #35471 (2021-05-20): k_heap: Clamp to a minimum heap size (and add test)
k_heap: Clamp to a minimum heap size

The K_HEAP_DEFINE macro would allow users to specify heaps that are
too small, leading to potential corruption events (though at least
there were __ASSERTs that would catch this at runtime if enabled).

It would be nice to put the logic to compute this value into the heap
code, but that isn't available in kernel.h (and we don't want to pull
it in as this header is already WAY to thick).  So instead we just
hand-compute and document the choice.  We can address bitrot problems
with a test.

Fixes #33009
commit d3737030 (2021-05-20): k_heap: Clamp to a minimum heap size

The K_HEAP_DEFINE macro would allow users to specify heaps that are
too small, leading to potential corruption events (though at least
there were __ASSERTs that would catch this at runtime if enabled).

It would be nice to put the logic to compute this value into the heap
code, but that isn't available in kernel.h (and we don't want to pull
it in as this header is already WAY to thick).  So instead we just
hand-compute and document the choice.  We can address bitrot problems
with a test.

(Tweaks to heap size asserts and correct size bounds from Nicolas Pi [... cut]
closed 2021-05-20 by nashif