zephyrproject-rtos/zephyr #33012: Support for the Raspberry Pi RP2040 soc

burumaj, 2021-03-06:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**
This request is for the support of RP2040 SoC by the Raspberry Pi foundation. This is a dual core
Cortex-M0+ device. The reason I think this would be a great addition to Zephyr is that it is a well documented SoC
with a well documented SDK too.

**Describe why you are asking for this support?**
Describe why you are asking for this support instead of contributing it directly to the tree

The reason I am asking for this support is because I am interested in using Zephyr on this chip.
However, I'm not that experienced with porting socs to zephyr, so i would probably need some guidance or help
taking on this task

Also seeing as Arduino, Adafruit and other distributors are creating boards for this soc.

If this is a new board or SoC, please state whether you are willing to maintain the Zephyr support for it if it is included in the main tree
If this SoC were to be included in the main tree i would be willing to maintain the code for it.

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the hardware here.

ioannisg, 2021-03-06:
CC @carlescufi

de-nordic, 2021-03-08:
Same issue here: https://github.com/zephyrproject-rtos/zephyr/issues/32193

ioannisg, 2021-03-09:
Draft PR started: #33172

voxpelli, 2021-07-24:
Not very helpful to have a bot close tracking issues like this one :/ Will only result in new ones being created and this adding to the noise rather than the intended opposite.

de-nordic, 2021-07-26:
There is active interest in the issue.

fkromer, 2021-07-31:
#37134

voxpelli, 2021-07-31:
> #37134

The Pico and the Compute Module are very different

fkromer, 2021-08-02:
@voxpelli Sorry. I wanted to reference another github issue.

voxpelli, 2021-10-02:
There is active interest in the issue.

(Any way to make the Stale bot ignore an issue?)

voxpelli, 2021-12-03:
Active interest

carlescufi, 2021-12-03:
The HAL request is here: #40817

PR #34835 (2022-02-02): boards: Add support for the raspberrypi pico
This PR provides very basic support for the Raspberry Pi Pico board. It continues #33172 by @ioannisg .
It currently supports the boot process and UART, and is able to run the `hello_world` sample.
As this is still very premature, it still lacks a few important features, such as spawning the second core or supporting `west flash`, but it's probably enough so that other people can start using this board with Zephyr and improve this port.
This PR assumes there is a `hal_raspberrypi` module under `zephyrproject-rtos`. Since I don't have permissions to  [... cut]
closed 2022-02-02 by carlescufi