zephyrproject-rtos/zephyr #33015: spi_nor driver: SPI_NOR_IDLE_IN_DPD breaks SPI_NOR_SFDP_RUNTIME

jrhees-cae, 2021-03-06:
Using the following configuration with the spi_nor flash driver:
CONFIG_SPI_NOR_SFDP_RUNTIME=y
CONFIG_SPI_NOR_IDLE_IN_DPD=y

This results in the flash initialization failing. This is due to the following sequence of events within the `spi_nor_configure()` function:

a) call to `spi_nor_read_jedec_id()`, which within calls `acquire_device` and `release_device`, thus leaving the flash in DPD mode
b) followed by a call to `spi_nor_process_sfdp()`, which in turn calls `read_sfdp()`, which in turn calls `spi_transceive()`

Thus, the call to `spi_transceive()` in step (b) above is made while flash chip is still in DPD mode. As such, the flash chip does not output SFDP data as expected.

pabigot, 2021-03-07:
#33103 outlines a way to resolve this using device power management, which is where DPD really belongs.

jasonxytan, 2022-09-07:
This issue is still not resolved. As described by the OP, the flash device will fail to initialize because the code does not acquire the flash (and exit DPD) when calling `read_sfdp()` in `spi_nor_process_sfdp()` [Line 880](https://github.com/zephyrproject-rtos/zephyr/blob/78d8ea4a6244279e3e866ec97543815d69f512e8/drivers/flash/spi_nor.c#L880) and [Line 913](https://github.com/zephyrproject-rtos/zephyr/blob/78d8ea4a6244279e3e866ec97543815d69f512e8/drivers/flash/spi_nor.c#L913), which results in invalid SFDP magic number.

Replacing `read_sfdp()` with `spi_nor_sfdp_read()` will resolve this.

The note for `read_sfdp()` actually says "The device must be externally acquired before invoking this function", which is  what `spi_nor_sfdp_read()` does. Is there any specific reason why `read_sfdp()` is used instead of `spi_nor_sfdp_read()`  here?

jasonxytan, 2022-09-07:
@jrhees-cae Would you mind reopening this issue? I think it would be better not to create a duplicated issue.

jrhees-cae, 2022-09-07:
Sorry I cannot re-open -- the author can only re-open if they themselves closed the issue.

jasonxytan, 2022-09-07:
Thank you @jrhees-cae. Would you mind reopening this issue @pabigot? Thank you.

pabigot, 2022-09-07:
@jasonxytan done, but be aware that if no action is taken it will automatically be closed again in sixty days, unless Zephyr has changed its clean-the-books policy since I was last involved.

PR #50111 (2022-09-14): drivers: flash: spi_nor: fix driver initialization error
Fix the issue that the initialization will fail when both CONFIG_SPI_NOR_SFDP_RUNTIME and CONFIG_SPI_NOR_IDLE_IN_DPD are enabled. The cause of this problem is simply calling the wrong function.

Fixes #33015
commit 904181a6 (2022-09-14): drivers: flash: spi_nor: fix driver initialization error

Fix the issue that the initialization will fail when both
CONFIG_SPI_NOR_SFDP_RUNTIME and CONFIG_SPI_NOR_IDLE_IN_DPD are enabled.
The cause of this problem is simply calling the wrong function.

Fixes #33015
closed 2022-09-14 by carlescufi