zephyrproject-rtos/zephyr #33016: spi_nor: CONFIG_SPI_NOR_SFDP_RUNTIME leaves flash in Standby after spi_nor_configure()

jrhees-cae, 2021-03-06:
Using the following configuration with the spi_nor flash driver:
CONFIG_SPI_NOR_IDLE_IN_DPD=y

and
   CONFIG_SPI_NOR_SFDP_DEVICETREE=y  OR  CONFIG_SPI_NOR_SFDP_MINIMAL=y 

This results in the the flash chip being left in STANDBY (not DPD) upon exit of `spi_nor_configure()`. In other words, the flash, after initialization of the spi_nor driver, is not in DPD. Subsequent flash accesses will correct this issue and leave the flash in DPD.

This is caused by lines 907-910 of `spi_nor.c`:
```
      if (IS_ENABLED(CONFIG_SPI_NOR_IDLE_IN_DPD)
          && (enter_dpd(dev) != 0)) {
              return -ENODEV;
      }
```
While this code is intended to put the FLASH chip into DPD, it actually causes the chip to EXIT DPD, since the flash was already placed in DPD previously in the function by a call to `spi_nor_read_jedec_id()`.

Since any activity of the CS line will cause the DPD mode to be exited (and the command sent while the chip is in DPD will be ignored), the call to `enter_dpd()` while the chip is already in DPD will cause the opposite of the intended effect.

pabigot, 2021-03-07:
#33103 outlines a way to resolve this using device power management, which is where DPD really belongs.

jrhees-cae, 2021-03-07:
Agreed that this needs to be moved to a PM layer. For example, in my application, there exists a thread that works with the flash directly (for app-driven custom storage), and then there is also an Image management component module of Zephyr that accesses the flash for storage of upgrade images for MCUBoot. Explicitly power-managing the flash from an app level is not sufficient, since there would not be coordination among all the users of the flash. However, extending the concept of acquire/release out to the zephyr flash API would allow the various users of flash to acquire and release flash at a less granular level. A refcount mechanism could then be used within the flash drivers to determine when to enter and exit DPD, etc.

closed 2021-06-06 by github-actions[bot]