zephyrproject-rtos/zephyr #33029: [Coverity CID :219660] Untrusted divisor in drivers/sensor/bme680/bme680.c

zephyrbot, 2021-03-07:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/drivers/sensor/bme680/bme680.c#L382

Category: Insecure data handling
Function: `bme680_chip_init`
Component: Drivers
CID: [219660](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=219660)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/drivers/sensor/bme680/bme680.c#L382

```
376         err = bme680_reg_write(data, BME680_REG_CTRL_GAS_1,
377                        BME680_CTRL_GAS_1_VAL);
378         if (err < 0) {
379             return err;
380         }
381    
>>>     CID 219660:  Insecure data handling  (TAINTED_SCALAR)
>>>     Passing tainted expression "data->res_heat_val" to "bme680_calc_res_heat", which uses it as a divisor or modulus.
382         err = bme680_reg_write(data, BME680_REG_RES_HEAT0,
383                        bme680_calc_res_heat(data, BME680_HEATR_TEMP));
384         if (err < 0) {
385             return err;
386         }
387    
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

MaureenHelm, 2021-03-09:
@BoschSensortec will you please take a look?

gmarull, 2021-05-21:
It turned out to be a false positive, so closing.

Details provided in the coverity issue:
```
Because var5 is used as a divisor, it could be a problem if it is zero. However, var5 will never be zero. It is the result of the expression:

var5 = (131 * data->res_head_val) + 65536.

The minimum value data->res_head_val can take is -128 (int8_t), which leads to var5 = -48768. Even if it was an int16_t or int32_t, 131 is not a divisor of 65536, so var5 would never be zero.
```

closed 2021-05-21 by gmarull