zephyrproject-rtos/zephyr #33038: [Coverity CID :219649] Operands don't affect result in kernel/sem.c

zephyrbot, 2021-03-07:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/kernel/sem.c#L68

Category: Integer handling issues
Function: `z_impl_k_sem_init`
Component: Kernel
CID: [219649](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=219649)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/kernel/sem.c#L68

```
62     int z_impl_k_sem_init(struct k_sem *sem, unsigned int initial_count,
63                   unsigned int limit)
64     {
65         /*
66          * Limit cannot be zero and count cannot be greater than limit
67          */
>>>     CID 219649:  Integer handling issues  (CONSTANT_EXPRESSION_RESULT)
>>>     "limit > 4294967295U /* 2147483647 * 2U + 1U */" is always false regardless of the values of its operands. This occurs as the logical second operand of "||".
68         CHECKIF(limit == 0U || limit > K_SEM_MAX_LIMIT || initial_count > limit) {
69             return -EINVAL;
70         }
71    
72         sem->count = initial_count;
73         sem->limit = limit;
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

dcpleung, 2021-03-07:
@jharris-intel is `limit > K_SEM_MAX_LIMIT` needed in this case? I see `K_SEM_MAX_LIMIT` is #define to be `UINT_MAX`.

jharris-intel, 2021-03-08:
What's the recommended approach when a limiting constant could be `UINT_MAX` in some, but not all, configurations?

That constant was added because I've got a WIP PR that adds an implementation whose limit is `INT_MAX`.

dcpleung, 2021-03-08:
I assume we can ignore it as intentional in Coverity? @ceolin what do you think?

ceolin, 2021-03-08:
> I assume we can ignore it as intentional in Coverity? @ceolin what do you think?

Yep, we have others similar cases. This just need to be triaged as intentional

dcpleung, 2021-03-08:
Done.

closed 2021-03-08 by dcpleung