zephyrproject-rtos/zephyr #33041: [Coverity CID :219645] Untrusted loop bound in subsys/bluetooth/host/sdp.c

zephyrbot, 2021-03-07:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/subsys/bluetooth/host/sdp.c

Category: Insecure data handling
Function: `bt_sdp_get_profile_version`
Component: Bluetooth
CID: [219645](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=219645)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/subsys/bluetooth/host/sdp.c#L2492

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

carlescufi, 2021-05-20:
@MaureenHelm assigning this to you since @MarkWangChinese is not yet part of the org, and this is BR/EDR with which you've shown an interest in keeping alive.

MaureenHelm, 2021-05-24:
@MarkWangChinese can you take a look this week?

MarkWangChinese, 2021-05-27:
I think it is not a issue, we can set it as "ignore":
Coverity think the "data length" is calculated by data of the net_buf, and the data came from an external source (we know it is BT).
Then Coverity think the "data length" is tainted and untrusted. Coverity want code to give the "data length" an permissible range, but the "data length" is calculated by the SDP string type (it's length has no permissible range).
The codes have check that the "data length" must be less than the remaining length of net_buf, I think it is the "permmissible range" that Coverity expects.

Or do we have any way to check the "permmissible range" of net_buf->len?

carlescufi, 2021-06-03:
@MarkWangChinese See my comment in #33052

carlescufi, 2021-06-17:
@MarkWangChinese ping

carlescufi, 2021-07-15:
@MarkWangChinese @MaureenHelm ping

cfriedt, 2021-09-23:
@dleach02  - can you please have a look at this?

PR #38821 (2021-09-27): Bluetooth: SDP: Check len is not zero before accessing data pointer
Check len is not zero before accessing data pointer, the len variable
is not checked before this point so cannot be trusted to not be zero.

FIxes: #33041
Fixes: #33052 
Fixes: #38116
closed 2021-09-27 by carlescufi