zephyrproject-rtos/zephyr #33052: [Coverity CID :219624] Untrusted loop bound in subsys/bluetooth/host/sdp.c

zephyrbot, 2021-03-07:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/subsys/bluetooth/host/sdp.c

Category: Insecure data handling
Function: `bt_sdp_get_proto_param`
Component: Bluetooth
CID: [219624](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=219624)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/subsys/bluetooth/host/sdp.c#L2468

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

carlescufi, 2021-05-20:
@MaureenHelm assigning this to you since @MarkWangChinese is not yet part of the org, and this is BR/EDR with which you've shown an interest in keeping alive.

MaureenHelm, 2021-05-24:
@MarkWangChinese can you take a look this week?

MarkWangChinese, 2021-05-27:
I think it is not a issue, we can set it as "ignore":
Coverity think the "data length" is calculated by data of the net_buf, and the data came from an external source (we know it is BT).
Then Coverity think the "data length" is tainted and untrusted. Coverity want code to give the "data length" an permissible range, but the "data length" is calculated by the SDP string type (it's length has no permissible range).
The codes have check that the "data length" must be less than the remaining length of net_buf, I think it is the "permmissible range" that Coverity expects.

Or do we have any way to check the "permmissible range" of net_buf->len?

carlescufi, 2021-06-03:
@MarkWangChinese this is actually a valid coverity bug. This part here:

[image]

assumes that there are at least 2 valid bytes pointed to by `data*` but in fact, the loop iterator (`while (data - buf->data < buf->len)`) only guarantees you one, and on top of that there is a increment of `data*` before the `sys_get_be16(data);`. You should be using `net_buf_pull_*` in general instead of accessing `data*` directly in order to have it checked.

carlescufi, 2021-06-17:
@MarkWangChinese ping

MarkWangChinese, 2021-06-21:
Hi @carlescufi you are right. In this API, it seems not want to change the buf->data, so the variable *data is used. I will add some "if" to the codes.
From my understand and the Coverity log, your found seems not related to the Coverity logic. Coverity think the  "data length" is tainted and untrusted because it comes from external source.
[image]

MarkWangChinese, 2021-07-12:
Hi @carlescufi I created one PR https://github.com/zephyrproject-rtos/zephyr/pull/36876, please check whether it is what you expect.

carlescufi, 2021-07-15:
@MarkWangChinese did #36876 fix this Coverity issue?

joerchan, 2021-08-02:
ping @MarkWangChinese

cfriedt, 2021-09-23:
@dleach02 - can you please have a look at this?

joerchan, 2021-09-24:
Since we didn't get a response from @MarkWangChinese I went and checked myself. To my understanding the fix already merged handles the issue (if that is what coverity is complaining about). I did however find another untrusted source, submitted another fix for that.

There are 3 coverity reports originating from the same code, which could probably benefit from a change to using the net_buf module to parse the incoming data.

@dleach02 I will mark these 3 reports as fixed, but I'm still not 100% sure that there won't be a new report. But to my understand of the code it is now correctly handled with both fixes.

PR #38821 (2021-09-27): Bluetooth: SDP: Check len is not zero before accessing data pointer
Check len is not zero before accessing data pointer, the len variable
is not checked before this point so cannot be trusted to not be zero.

FIxes: #33041
Fixes: #33052 
Fixes: #38116
closed 2021-09-27 by carlescufi