zephyrproject-rtos/zephyr #33053: [Coverity CID :219623] Out-of-bounds read in drivers/ipm/ipm_nrfx_ipc.c

zephyrbot, 2021-03-07:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/drivers/ipm/ipm_nrfx_ipc.c

Category: Memory - illegal accesses
Function: `vipm_dispatcher`
Component: Drivers
CID: [219623](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=219623)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/drivers/ipm/ipm_nrfx_ipc.c#L135

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

masz-nordic, 2021-04-06:
False positive, `event_idx` will never exceed `NRFX_IPC_ID_MAX_VALUE`.

closed 2021-04-06 by masz-nordic

ioannisg, 2021-04-07:
@masz-nordic did you mark the issues as false positive in Coverity as well?

masz-nordic, 2021-04-07:
@ioannisg Yes I did.