zephyrproject-rtos/zephyr #33073: [Coverity CID :219591] Untrusted divisor in drivers/sensor/bme680/bme680.c

zephyrbot, 2021-03-07:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/drivers/sensor/bme680/bme680.c

Category: Insecure data handling
Function: `bme680_sample_fetch`
Component: Drivers
CID: [219591](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=219591)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/drivers/sensor/bme680/bme680.c#L224

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

MaureenHelm, 2021-03-09:
@BoschSensortec will you please take a look?

MaureenHelm, 2021-03-29:
@BoschSensortec ping

galak, 2021-05-13:
Looked the details here and the divisor is ranged to 0..15 by mask of BME680_MSK_GAS_RANGE.  So closing this as a false positive.

closed 2021-05-13 by galak