zephyrproject-rtos/zephyr #33093: [Coverity CID :215391] Unchecked return value from library in samples/net/mdns_responder/src/service.c

zephyrbot, 2021-03-07:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/samples/net/mdns_responder/src/service.c

Category: Error handling issues
Function: `welcome`
Component: Samples
CID: [215391](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=215391)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/bd97359a5338b2542d19011b6d6aa1d8d1b9cc3f/samples/net/mdns_responder/src/service.c#L26

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

GunZi200, 2021-03-07:
@pfalcon @jukkar @tbursztyka I see you are assigned to this, what do you think of my proposed solution below? 😃 

I don't think a simple `(void)` cast is a good solution here.

From here I can see information about the return value: https://pubs.opengroup.org/onlinepubs/9699919799/functions/send.html

> Upon successful completion, send() shall return the number of bytes sent. Otherwise, -1 shall be returned and errno set to indicate the error.

So my solution would be something like this:

```c
static void welcome(int fd)
{
    static const char msg[] = "Bonjour, Zephyr world!\n";

    int r = send(fd, msg, sizeof(msg), 0);
    if (r == -1) {
        NET_DBG("send() failed (%d)", errno);
        close(fd);
    }
}
```

I would also change the return type of `welcome()` from `void` to `int` so that we can terminate the process in `service()`. Where `-1` indicates error, and `0` indicates success.

```c
void service(void)
{
    // ... some code above ...
    /* send a banner */
    if(welcome(client_fd) == -1) {
        return;
    }
   // ... some code below ...
}
```

```c
static int welcome(int fd)
{
    static const char msg[] = "Bonjour, Zephyr world!\n";

    int r = send(fd, msg, sizeof(msg), 0);

    if (r == -1) {
        NET_DBG("send() failed (%d)", errno);
        close(fd);
        return r;
    }

    return 0;
}
```

jukkar, 2021-03-08:
Hi @GunZi200, your proposal looks fine. Can you send a PR for it?

After looking the application, it tends to quit far too easy. So it works only once, and quits for any errors. It would need some more work to be more usable. Anyway, that is for another PR.

jukkar, 2021-03-08:
@GunZi200 If you are at it, could you also fix #33094 which leaks server_fd, by not closing it if there is an error in bind() or listen() in service.c. Thanks in advance!

GunZi200, 2021-03-08:
Sure I’ll create a PR for both issues this evening :)

PR #33162 (2021-03-09): samples: net: mdns_responder: fix coverity issues
* Fix #33093 [Coverity CID :215391] Unchecked return value from library in `samples/net/mdns_responder/src/service.c`
* Fix #33094 [Coverity CID :215381] Resource leak in `samples/net/mdns_responder/src/service.c`
commit 7713d347 (2021-03-09): samples: net: mdns_responder: Fix coverity issue

Check return value if sendto() fails.

Coverity-CID: 215391
Fixes #33093
closed 2021-03-09 by jukkar