zephyrproject-rtos/zephyr #33100: tcp2 not working with ppp

shlomow, 2021-03-07:
**Describe the bug**
When running ppp and tcp with the new tcp stack, it does not work.
After switching to the legacy tcp stack everything worked as expected.

**To Reproduce**
Build the ppp support with the network shell support. The default tcp stack is tcp2.
Then, try to connect to a server using `net tcp connect` command.

jukkar, 2021-03-08:
More information is needed.
What version (upstream, 2.5 or what) are you using?
How does it not work, connect fails or data transfer fails or what exactly is failing?

I do not have a SIM card so cannot try this myself with gsm modem.

shlomow, 2021-03-08:
I am using the latest master commit. You do not need the modem reproduce the issue. Just enable ppp in your configuration. From what I saw no tcp packet exit the target device

jukkar, 2021-03-08:
> You do not need the modem reproduce the issue. Just enable ppp in your configuration.

Yep, I realized that too. I have a test environment now ready.

jukkar, 2021-03-08:
Hmm, at least a simple tcp connect/send test to Linux works ok. In linux side I have
```
nc -l 192.0.2.2 4242
```
and when connecting to it from zephyr
```
uart:~$ net tcp connect 192.0.2.2 4242
Connecting from 192.0.2.1:0 to 192.0.2.2:4242
uart:~$ net tcp send bar
Message sent
```
I am seeing data in the linux terminal. Also receiving works.

Could you elaborate in more details how to replicate this issue?

jukkar, 2021-03-08:
BTW, I used following config in Zephyr side (need three terminals):
```
$ cd net-tools
$ ./loop-ppp-dev.sh
```
```
$ cd net-tools
$ sudo ./loop-pppd.sh
```
```
$ cd samples/net/sockets/echo_server
$ cmake . -Bbuild -DBOARD=qemu_x86 -DOVERLAY_CONFIG=overlay-ppp.conf -DCONFIG_NET_PPP_UART_NAME=\"UART_1\"
$ make -Cbuild run
```

mniestroj, 2021-03-08:
I guess the problem is not with PPP itself, but with `net tcp connect` command. This command uses non-blocking `connect()`, which is not supported by TCP2. In TCP case the "non-blocking" request was still "blocking", which is why it worked.

jukkar, 2021-03-08:
I was using TCP2 with net-shell and it worked just fine. The commit 9eb46d95a7068b6fa3cb76c4625b8efcbd58bbfa made it possible to use TCP2 in net-shell.

closed 2021-03-12 by shlomow