zephyrproject-rtos/zephyr #33108: doc: Adding documentation for porting a new SoC

burumaj, 2021-03-07:
**Is your enhancement proposal related to a problem? Please describe.**
Missing documentation on porting a new SoC to Zephyr.  The lack of of this documentation
poses a significant hurdle for people not super familiar with this part of the process.

**Describe the solution you'd like**
The solution I would like to see is porting guide for a new SoC similar to the porting guides
for new architectures, boards and shields (located [here](https://docs.zephyrproject.org/latest/guides/porting/index.html)).

As far as I can tell for now, this should at least include where Kconfig looks for certain configuration options.
For example to select a new family of SoC Kconfig sources `soc/<arch>/<family>/Kconfig.soc`.

**Describe alternatives you've considered**
Right now the only alternative i can see is the laborious process of following the way other families and series of 
SoC's are ported. This is made more difficult since not every family and series follows the same naming conventions.

**Additional context**
This was all encountered while working on a SoC port for the RP2040

burumaj, 2021-03-07:
There is more information already compiled in this presentation:
https://events19.linuxfoundation.org/wp-content/uploads/2017/12/Enabling-Zephyr-on-Your-Hardware-Platform-Diego-Sueiro-Sepura-Embarcados.pdf

K2AAE, 2021-03-08:
Dead end: "This presentation will not cover how to add a new CPU core architecture support. But a good documentation on how to achieve this can be find here;"  and "here" takes you to https://docs.zephyrproject.org/latest/porting/arch.html which returns Sorry, Page Not Found. I've found hunting down answers on Zephyr to be endless loops, nRF stuff a good example of lazy.

burumaj, 2021-03-11:
The docs for porting a new architecture can be found here (https://docs.zephyrproject.org/latest/guides/porting/arch.html). This is more related to adding a new SoC based on an already supported architecture.

GunZi200, 2021-03-24:
I find the documentation here (in the board porting guide) could be improved for new users (like myself):

> Start by making sure your SoC is supported by Zephyr

I think it will be helpful for everyone this at least include some links here if some exist to help save time. 
[image]