zephyrproject-rtos/zephyr #33120: iotdk: running testcase tests/kernel/mbox/mbox_api/ failed

Zhaoningx, 2021-03-08:
To Reproduce

I ran testcase tests/kernel/mbox/mbox_api/ failed

Steps to reproduce the behavior:

twister -p iotdk --device-testing --device-serial=/dev/ttyUSB0 -T tests/kernel/mbox/mbox_api/  -j 1 -vv

see error

===================================================================
START - test_mbox_async_put_to_waiting_get
PASS - test_mbox_async_put_to_waiting_get

===================================================================
START - test_mbox_get_waiting_put_incorrect_tid
PASS - test_mbox_get_waiting_put_incorrect_tid

===================================================================
START - test_mbox_async_multiple_put
PASS - test_mbox_async_multiple_put

===================================================================
START - test_mbox_data_get_null
E: ***** Exception vector: 0x6, cause code: 0x2, parameter 0x2
E: Address 0x80000824
E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
E: Current thread: 0x80000168 (unknown)
E: Halting system

Environment
OS: Fedora33
Toolchain:zephyr-sdk-0.12.2
Commit ID: 7b8bce898fd983595ae7043

galak, 2021-03-16:
#33121

galak, 2021-03-16:
Fixed by #33121

closed 2021-03-16 by galak