zephyrproject-rtos/zephyr #33123: API extensions for Cache Management

nika-nordic, 2021-03-08:
## Introduction

We would like to propose several extensions to Zephyr’s Cache Management API.

### Problem description

Some SoC may implement extra cache features that are not covered by current Zephyr cache API.

### Proposed change

Add new functions to existing Zephyr cache API that would help to utilize miscellaneous cache capabilities.

## Detailed RFC

### Proposed change (Detailed)

Expand Zephyr cache management API with the following functionalities:

- cache contents erase
•	for security reasons we may need to perform cache erase operations
- cache status check (for SoCs that do not halt the CPU during cache operations)
- cache save & restore
•	for deep-sleep retention purposes
•	this might relate to power management API as well.
- cache profiling (for application performance measurements)

### Dependencies

This would be a new API so current code is unaffected.

ioannisg, 2021-03-11:
CC @ceolin @andyross

decsny, 2023-08-21:
Another thing that would be useful is if the cache API supported platforms that have multiple caches, for example some NXP platforms have cache64 which is paired with the flexspi to fetch from external memories (so, it's kind of like an L2 cache, but specific only to one peripheral), and also the core cache for the instruction and data buses generally with any bus followers