zephyrproject-rtos/zephyr #33127: Improve documentation user experience

gmarull, 2021-03-08:
## Introduction

The aim of this RFC is to trigger a discussion on possible improvements to the Zephyr documentation user experience (UX). These improvements could be theme style enhancements, toctree structure, etc.

### Problem description

While Zephyr documentation has lots of useful resources I think it requires some attention regarding usability. While it is a very subjective topic, I think we can compare with other open source projects and see what we can learn.

[image]
_Example of a more curated documentation (Godot) vs. Zephyr_

### Proposed change

- Discuss possible theme style enhancements. Some ideas:
  - Customize `sphinx-rtd-theme` further, e.g. see https://docs.godotengine.org/en/latest/index.html
  - Create a new Sphinx template?
- Discuss toctree layout. Splitting into multiple toctrees with its own `caption` may enhance navigation and visibility, e.g. see https://docs.godotengine.org/en/latest/index.html
- Create a "Tutorial" focused new section. While current samples are a good source of how-to use an API or a device, they frequently miss contextual information, e.g. Why enable this Kconfig option, why changing this priority...?. These guides could be a good starting point for new or less experienced developers.
- Improve search experience (see #32997)

## Detailed RFC

TBD after initial discussions.

### Proposed change (Detailed)

TBD after initial discussions.

### Dependencies

N/A

### Concerns and Unresolved Questions

N/A

## Alternatives

N/A

wldevries, 2021-03-08:
It would be nice to have an Edit button at the top that brings you to the source of the manual page. It may help raise problems with documentation sooner and bring in contributors.

See for example the open source documentation on the Microsoft website:
https://docs.microsoft.com/en-us/dotnet/csharp/language-reference/builtin-types/integral-numeric-types

wldevries, 2021-03-10:
Something else I'd like to see as a newbie to Zephyr are links from the documentation of a sample to the source code of the samples in git, or maybe embedded into the documentation as content.

carlescufi, 2021-03-11:
dev-review:

>Discuss possible theme style enhancements. Some ideas:
>Customize sphinx-rtd-theme further, e.g. see https://docs.godotengine.org/en/latest/index.html
>Create a new Sphinx template?

Create a new .css file and prototype/publish to cover this item.
Marketing color/logo guidelines here: https://drive.google.com/drive/u/0/folders/0B8CLJ6f1yrKEMmpaWkItMFhEc28

gregshue, 2021-03-11:
As we consider improving the documentation experience, pay special attention to and documents are generated and served for:

- the T2: Star - and T3: Forest topologies (where Zephyr is not the manifest repository)
- security artifacts
- a private, modified fork of Zephyr

While working in private external modules I have needed to verify documentation in portions of the module are automatically picked up when generating Zephyr documentation. I have also found it very helpful to highly leverage the Zephyr document generation solution and document organization for my manifest repository (which happens to also be a module). I consider being able to do this at no cost and without external services to be a very valuable part of the Zephyr ecosystem.

cfriedt, 2021-03-11:
I'd really like to see some documentation guidelines for external modules. In particular, because Zephyr's generated doc's are so nice, it would be great if modules could get something similar at near zero cost (i.e. very easily).

E.g. I know that there is a module template repository on GitHub, so if some of the documentation tooling was added there, it would be great.

gregshue, 2021-03-13:
Please anticipate:
- a manifest repository may contain multiple manifests (e.g. one per zephyr fork being supported by the workspace)
- an external module repository may also contain build files, glue code, and documentation for other ecosystems, i.e.
  - documentation discovery must follow `$myModule/zephyr/module.yml`
  - documentation tools must not require necessary files to be in the root of any repository
- building documentation for just the repository (module) or for the entire workspace (manifest)
- needing the document generation process and generated documentation to be fully functional without any Internet access
- needing to support subsystem design docs in addition to overview/integration docs
- needing to support a  tree of product/project configurations analogous to `tests/` and `samples/`

gmarull, 2021-03-14:
@gregshue Current improvements are geared towards user experience improvements, so build infrastructure / options should not be affected. Regarding internet connection: if we end up using a server side search for the official docs the default Sphinx search will still be kept. Thanks for the detailed list by the way!

gregshue, 2021-03-15:
> Discuss toctree layout. Splitting into multiple toctrees with its own caption may enhance navigation and visibility, e.g. see https://docs.godotengine.org/en/latest/index.html

Most of my list above depends on toctrees improvements to support the **scalability** of the current modular build system (hence "anticipate").  Once we have tools and guidelines for how to organize and navigate content at scale, populating the content it straight-forward.

gmarull, 2021-03-15:
@gregshue multiple toctrees should not affect the build system, it's only about doing the split on the rst side so that we have something like this in the sidebar:

[image]

gregshue, 2021-03-15:
@gmarull Understood. I have already made a parallels of the Zephyr document generation system in my draft zephyr module template (https://github.com/gregshue/oot_module_example) and added some rst content into my module. I have already added my module into my local clone of the Zephyr repository and built Zephyr documentation, seeing some of my documentation already appear in the generated Zephyr html files.

What I learned was the Zephyr sidebar (and toctrees layout) is not organized/layered well enough present/reflect the modularity and scalability **already present** in the Zephyr build system.  I am providing feedback that the toctrees need to be split/layered/reorganized to reflect the modularity that exists so that the navigation experience flows as the content scales up.

gmarull, 2021-05-17:
Closing now that new theme is in place. More specific issues can be opened to track further improvements.

closed 2021-05-17 by gmarull