zephyrproject-rtos/zephyr #33147: Not able to build blinky or set toolchain to zephyr

moose-rivieh, 2021-03-08:
**Describe the bug**
Not able to build the blinky example by following the instructions listed in https://docs.zephyrproject.org/latest/getting_started/index.html
I get the following error when I get to the west build command:

```
-- west build: generating a build system
Including boilerplate (Zephyr base (cached)): /home/mustafa/Documents/zephyr/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/mustafa/Documents/zephyr/zephyr/samples/basic/blinky
-- Zephyr version: 2.5.99 (/home/mustafa/Documents/zephyr/zephyr)
-- Found west (found suitable version "0.9.0", minimum required is "0.7.1")
-- Board: bl653_dvk
-- Cache files will be written to: /home/mustafa/.cache/zephyr
-- Using toolchain: zephyr 0.12.3 (/home/mustafa/.local/zephyr-sdk-0.12.3)
-- Found dtc: /usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found toolchain: gnuarmemb (/home/mustafa/Programs/gnuarmemb)
CMake Error at /home/mustafa/Documents/zephyr/zephyr/cmake/compiler/gcc/generic.cmake:8 (message):
  Zephyr was unable to find the toolchain.  Is the environment misconfigured?

  User-configuration:

  ZEPHYR_TOOLCHAIN_VARIANT: gnuarmemb

  Internal variables:

  CROSS_COMPILE: /home/mustafa/Programs/gnuarmemb/bin/arm-none-eabi-

  TOOLCHAIN_HOME: /home/mustafa/Programs/gnuarmemb

Call Stack (most recent call first):
  /home/mustafa/Documents/zephyr/zephyr/cmake/generic_toolchain.cmake:42 (include)
  /home/mustafa/Documents/zephyr/zephyr/cmake/app/boilerplate.cmake:551 (include)
  /home/mustafa/Documents/zephyr/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /home/mustafa/Documents/zephyr/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:40 (include_boilerplate)
  CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!

FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/home/mustafa/Documents/zephyr/zephyr/build -S/home/mustafa/Documents/zephyr/zephyr/samples/basic/blinky -GNinja
```

The frustrating part is I don't know why it is using gnuarmemb toolchain?? I explicitly set ZEPHYR_TOOLCHAIN_VARIANT to zephyr. Do not know what to do next?

**To Reproduce**
```
cd ~/Documents/zephyr/zephyr
export ZEPHYR_TOOLCHAIN_VARIANT=zephyr
export ZEPHYR_SDK_INSTALL_DIR=~/.local/zephyr-sdk-0.12.3
source zephyr-env.sh
west build -p auto -b bl653_dvk samples/basic/blinky
```
**Expected behavior**
Build blinky sample example for bl653 dev board using zephyr toolchain.

**Impact**
showstopper

**Environment (please complete the following information):**
Ubuntu 20.04

carlescufi, 2021-03-08:
It might be that you have a `~/.zephyrrc` that sets the the toolchain variant to `zephyr` and that is parsed and executed when you run `source zephyr-env.sh`.

moose-rivieh, 2021-03-08:
> It might be that you have a `~/.zephyrrc` that sets the the toolchain variant to `zephyr` and that is parsed and executed when you run `source zephyr-env.sh`.

My goal is to set the toolchain to zephyr but for some reason, it is pointing to gnuarmemb. Please see the console output that I copied above. In either case, there is no .zephyrrc file in my home directory.

nashif, 2021-03-08:
what do you have in ~/.zephyrrc?

moose-rivieh, 2021-03-08:
> what do you have in ~/.zephyrrc?

I don't have a .zephyrrc file in my home directory. Do I need to create one to solve this issue? If I do, what content do I need to include?

moose-rivieh, 2021-03-09:
Any ideas here?

carlescufi, 2021-03-09:
@moose-rivieh Not really, no. When you execute `source zephyr-env.sh`, that script will read a `~/.zephyrrc` but only if it exists, and it's not necessary to have it at all. Something is setting `ZEPHYR_TOOLCHAIN_VARIANT` in your system, and it certainly is not Zephyr if you don't have a `~/.zephyrrc`. Please check your own scripts or system for anything setting that variable.

moose-rivieh, 2021-03-09:
Yeah, that is wired. I don't believe I have other things that could be setting it. I don't know how exactly environment variables work, but I thought by calling export ZEPHYR_TOOLCHAIN_VARIANT=zephyr, I am explicitly setting that variable to zephyr regardless if other systems are setting it differently. Is there a more bulletproof way to set this variable? Is there a file that I can manually edit? Is it possible it is a bug with the west tool? Has anybody been able to switch ZEPHYR_TOOLCHAIN_VARIANT back and forth between zephyr and gnuarmemb? I am assuming so, but I can't...

carlescufi, 2021-03-10:
@moose-rivieh please make sure to clean up your `zephyr/` folder completely from any `build/` folders you might have and try again. You can run `git clean -fdx` (careful, this will erase all untracked files) to make sure.

carlescufi, 2021-03-18:
Closing for lack of activity. Please reopen if applicable.

closed 2021-03-18 by carlescufi