zephyrproject-rtos/zephyr #33149: subsys: canbus: canopen EDSEditor / libedssharp version that works with Zephyr's CANopenNode

Qbicz, 2021-03-08:
**Describe the bug**
Zephyr maintains its own fork of CANopenNode https://github.com/zephyrproject-rtos/canopennode/commits/zephyr.

To generate .c and .h CO_OD files that contain description of the CAN device, libedssharp is used. However:
it's most recent release 0.8 is **not compatible** with the Zephyr's version (too old) https://github.com/robincornelius/libedssharp/releases -> it gives error like here https://github.com/zephyrproject-rtos/zephyr/issues/26363
- Latest master is also not compatible (too new): it gives error `zephyrproject/modules/lib/canopennode/stack/CO_TIME.h:107:5: error: unknown type name 'TIME_OF_DAY'`

I noticed that the sample files in Zephyr repo were generated with version 0.8.7: https://github.com/zephyrproject-rtos/zephyr/blame/master/samples/subsys/canbus/canopen/objdict/CO_OD.h#L13
Newer / older version seem to be incompatible, do you have the binary / exe available?

I tried to build the libedssharp with mono/xbuild (Mono Version 6.8.0.105) on Linux, but it looks like the mono compiler lacks a few features that are used in this C# program: `DeviceODView.cs(118,48): error CS1644: Feature 'default literal' cannot be used because it is not part of the C# 7.0 language specification`
That's why I ask for a binary. Alternatively please point me to an instruction on how to build the https://github.com/robincornelius/libedssharp on Linux. I'm a C# noob.

I found the project CI, however the artifacts are not available I think: https://travis-ci.org/github/robincornelius/libedssharp/jobs/746239909

Any idea @henrikbrixandersen?

Qbicz, 2021-03-22:
Any update, or suggestion how should I approach this?

henrikbrixandersen, 2021-03-25:
> **Describe the bug**
> Zephyr maintains its own fork of CANopenNode https://github.com/zephyrproject-rtos/canopennode/commits/zephyr.
>
> To generate .c and .h CO_OD files that contain description of the [...]

The latest upstream release of libedssharp (v0.8) unfortunately contains a bug regarding the name of the CO_NO_TIME/CO_NO_TS definition. It was fixed a few commits after upstream v0.8 was released (in https://github.com/robincornelius/libedssharp/commit/2e53bbcb7c012805f02c7b9336efc67c743a95b1).

> * Latest master is also not compatible (too new): it gives error `zephyrproject/modules/lib/canopennode/stack/CO_TIME.h:107:5: error: unknown type name 'TIME_OF_DAY'`

The `master` branch of upstream libedssharp is dead and no longer used for active development as noted in the last commit to that branch: https://github.com/robincornelius/libedssharp/commit/cad4c74d70c09a1616761643d1f13dbc73d8e531
 
> I noticed that the sample files in Zephyr repo were generated with version 0.8.7: https://github.com/zephyrproject-rtos/zephyr/blame/master/samples/subsys/canbus/canopen/objdict/CO_OD.h#L13

Not v0.8.7, but v0.8-7 (v0.8 tag + 7 commits). This is the commit referenced above.

> Newer / older version seem to be incompatible, do you have the binary / exe available?

Unfortunately, I cannot provide binaries for this. You could try contacting upstream.

> I tried to build the libedssharp with mono/xbuild (Mono Version 6.8.0.105) on Linux, but it looks like the mono compiler lacks a few features that are used in this C# program: [...]

I don't have any issues building libedssharp on Ubuntu 20.04 LTS using Mono v6.8.0.105.

```
sudo apt-get install -y mono-xbuild mono-complete libcanberra-gtk-module
xbuild EDSTest/EDSEditorGUI.csproj

```

Qbicz, 2021-03-25:
Hi @henrikbrixandersen, thank you for your detailed answer.

> The latest upstream release of libedssharp (v0.8) unfortunately contains a bug regarding the name of the CO_NO_TIME/CO_NO_TS definition. It was fixed a few commits after upstream v0.8 was released (in robincornelius/libedssharp@2e53bbc).

I ended up with using a latest release libedssharp **v0.8** + a small definition in Zephyr's project CMakeLists to handle the problematic CO_NO_TIME/CO_NO_TS define:

`zephyr_compile_definitions(CO_NO_TIME=CO_NO_TS)`

Big thank you to @pawelzadrozniak for this suggestion.

closed 2021-03-25 by Qbicz