zephyrproject-rtos/zephyr #33153: west flash cannot find OpenOCD

glenn-andrews, 2021-03-08:
**Describe the bug**
west flash fails when trying to use OpenOCD to flash an STM32F401RE Nucleo-64 board
I have set the path as described in the documentation, and can run openocd --version from the command line and it returns the correct response.

**To Reproduce**
Steps to reproduce the behavior:
1. Follow the [getting started guide](https://docs.zephyrproject.org/latest/getting_started/index.html)
2. Download the OpenOCD files and set the path as stated [here](https://docs.zephyrproject.org/latest/guides/flash_debug/host-tools.html)
3. Build a sample app
4. type in west flash on the command line

**Expected behavior**
Flashing the Nucleo board, or an error saying why it's not possible to do so.

**Impact**
Showstopper. I can download .bin files using the ST-Link "copy to usb drive" function and they flash and run, but I am unable to debug.

**Logs and console output**
This is rebuilding the 'hello world' sample instead of blinky:

```
C:\Users\andrewsglenn\zephyrproject\zephyr>west flash
-- west flash: rebuilding
[1/2] Linking C executable zephyr\zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       13656 B       512 KB      2.60%
            SRAM:        4512 B        96 KB      4.59%
        IDT_LIST:          0 GB         2 KB      0.00%

[1/2] cmd.exe /C "cd /D C:\Users\andrewsglenn\zephyrprojec...\flash && "C:\Program Files\CMake\bin\cmake.exe" -E echo "

-- west flash: using runner openocd
Traceback (most recent call last):
  File "c:\python36\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\python36\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Python36\Scripts\west.exe\__main__.py", line 9, in <module>
  File "c:\python36\lib\site-packages\west\app\main.py", line 779, in main
    app.run(argv or sys.argv[1:])
  File "c:\python36\lib\site-packages\west\app\main.py", line 106, in run
    self.run_command(argv)
  File "c:\python36\lib\site-packages\west\app\main.py", line 338, in run_command
    self.run_extension(args.command, argv)
  File "c:\python36\lib\site-packages\west\app\main.py", line 408, in run_extension
    command.run(args, unknown, self.topdir, manifest=self.manifest)
  File "c:\python36\lib\site-packages\west\commands.py", line 129, in run
    self.do_run(args, unknown)
  File "C:\Users\andrewsglenn\zephyrproject\zephyr\scripts\west_commands\flash.py", line 29, in do_run
    do_run_common(self, my_args, runner_args)
  File "C:\Users\andrewsglenn\zephyrproject\zephyr\scripts/west_commands\run_common.py", line 222, in do_run_common
    runner.run(command_name)
  File "C:\Users\andrewsglenn\zephyrproject\zephyr\scripts/west_commands\runners\core.py", line 482, in run
    self.do_run(command, **kwargs)
  File "C:\Users\andrewsglenn\zephyrproject\zephyr\scripts/west_commands\runners\openocd.py", line 114, in do_run
    self.require(self.openocd_cmd[0])
  File "C:\Users\andrewsglenn\zephyrproject\zephyr\scripts/west_commands\runners\core.py", line 502, in require
    if shutil.which(program) is None:
  File "c:\python36\lib\shutil.py", line 1113, in which
    if os.path.dirname(cmd):
  File "c:\python36\lib\ntpath.py", line 242, in dirname
    return split(p)[0]
  File "c:\python36\lib\ntpath.py", line 204, in split
    p = os.fspath(p)
TypeError: expected str, bytes or os.PathLike object, not NoneType
```

**Environment (please complete the following information):**
 - OS: Windows 10 Enterprise
 - Latest Arm Tools and Zephyr downloaded on Friday.

mbolivar-nordic, 2021-03-09:
@glenn-andrews can you share the file `build/zephyr/runners.yaml` (where `build` is your build directory).

It should contain something like this:

```yaml

# Common runner configuration values.
config:
  ...
  # Host tools:
  ...
  openocd: /home/mbolivar/bin/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/openocd
  openocd_search: /home/mbolivar/bin/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts
```

If it doesn't, then cmake can't find your openocd for some reason. Your `build/CMakeCache.txt` file should contain a line like this:

```
//Path to a program.
OPENOCD:FILEPATH=/home/mbolivar/bin/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/openocd
```

That's what west uses. If CMake doesn't find it, west won't know about it.

The search itself is done here:

https://github.com/zephyrproject-rtos/zephyr/blob/33c9be90cceae0d8819431d2e01c6da221f7681e/cmake/host-tools.cmake#L48

Try debugging from there if you don't get the expected output in runners.yaml and CMakeCache.txt.

I don't think this is likely to be a west bug, so marking this as a 'question' until further information is provided.
I doubt this is actually a bug, but is more likely a misconfi

mbolivar, 2021-03-09:
Hi again. I can't reproduce this. The repro steps say:

> Build a sample app

I'm not sure what this means exactly since there are multiple ways to build an application.

The [board docs](https://docs.zephyrproject.org/latest/boards/arm/nucleo_f401re/doc/index.html) say you can use this:

```
west build -b nucleo_f401re samples/hello_world
```

Which is the usual way. I tried this on my Linux machine and got a runners.yaml as expected. I'm not sure what's going on in your setup but this doesn't look like either a west or a build system issue. Transferring assignee to @erwango since it's an STM32 board.

mbolivar-nordic, 2021-03-09:
(Reposting from the right GitHub account)

Hi again. I can't reproduce this. The repro steps say:

> Build a sample app

I'm not sure what this means exactly since there are multiple ways to build an
application.

The [board docs](https://docs.zephyrproject.org/latest/boards/arm/nucleo_f401re/doc/index.html) say you can use this:

```
west build -b nucleo_f401re samples/hello_world
```

Which is the usual way. I tried this on my Linux machine and got a
runners.yaml as expected. I'm not sure what's going on in your setup but this
doesn't look like either a west or a build system issue. Transferring assignee
to @erwango since it's an STM32 board.

glenn-andrews, 2021-03-09:
Sorry for the ambiguity. I used the build method in the Getting Started guide.

I tried again with a different sample using the following command:

west build -p auto -b nucleo_f401re samples\subsys\shell\shell_module

Once I did that, "west flash" operated as expected.

My guess is since I built it before openocd was installed, it wouldn't build the runners.yaml file until "-p auto" was provided.

closed 2021-03-10 by erwango