zephyrproject-rtos/zephyr #33168: CONFIG_HEAP_MEM_POOL_SIZE=64 doesn't work

yashi, 2021-03-09:
**Describe the bug**
Running `tests/kernel/mem_heap/mheap_api_concept` with `CONFIG_HEAP_MEM_POOL_SIZE=64` failed.

**To Reproduce**
```
$ cd zephyr
$ git describe
zephyr-v2.5.0-969-g2acec93ba6
$ cd ..
$ west update
```

this works
```
$ west build -p -b qemu_cortex_m3 -t run zephyr/tests/kernel/mem_heap/mheap_api_concept
```

this doesn't
```
$ west build -p -b qemu_cortex_m3 -t run zephyr/tests/kernel/mem_heap/mheap_api_concept -- -DCONFIG_HEAP_MEM_POOL_SIZE=64
```
Full output is placed below.

Interestingly `CONFIG_HEAP_MEM_POOL_SIZE` equals to 184 works but 183 doesn't.

**Expected behavior**
Expect the test to pass with 64 if there is no lower limit.

**Logs and console output**
```
$ west build -p -b qemu_cortex_m3 -t run zephyr/tests/kernel/mem_heap/mheap_api_concept -- -DCONFIG_HEAP_MEM_POOL_SIZE=64
-- west build: making build dir /home/yashi/src/zephyr-project/build pristine
-- west build: generating a build system
Including boilerplate (Zephyr base): /home/yashi/src/zephyr-project/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/yashi/src/zephyr-project/zephyr/tests/kernel/mem_heap/mheap_api_concept
-- Zephyr version: 2.5.99 (/home/yashi/src/zephyr-project/zephyr)
-- Found Python3: /home/yashi/src/zephyr-project/.venv/bin/python3.9 (found suitable exact version "3.9.1") found components: Interpreter 
-- Found west (found suitable version "0.9.0", minimum required is "0.7.1")
-- Board: qemu_cortex_m3
-- Cache files will be written to: /home/yashi/.cache/zephyr
-- Using toolchain: zephyr 0.12.3 (/home/yashi/zephyr-sdk-0.12.3)
-- Found dtc: /home/yashi/zephyr-sdk-0.12.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/yashi/src/zephyr-project/zephyr/boards/arm/qemu_cortex_m3/qemu_cortex_m3.dts
-- Generated zephyr.dts: /home/yashi/src/zephyr-project/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/yashi/src/zephyr-project/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/yashi/src/zephyr-project/build/zephyr/include/generated/device_extern.h
Parsing /home/yashi/src/zephyr-project/zephyr/Kconfig
Loaded configuration '/home/yashi/src/zephyr-project/zephyr/boards/arm/qemu_cortex_m3/qemu_cortex_m3_defconfig'
Merged configuration '/home/yashi/src/zephyr-project/zephyr/tests/kernel/mem_heap/mheap_api_concept/prj.conf'
Merged configuration '/home/yashi/src/zephyr-project/build/zephyr/misc/generated/extra_kconfig_options.conf'
Configuration saved to '/home/yashi/src/zephyr-project/build/zephyr/.config'
Kconfig header saved to '/home/yashi/src/zephyr-project/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/yashi/zephyr-sdk-0.12.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /home/yashi/src/zephyr-project/build
-- west build: running target run
[1/131] Preparing syscall dependency handling

[123/131] Linking C executable zephyr/zephyr_prebuilt.elf

[130/131] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       23444 B       256 KB      8.94%
            SRAM:        6248 B        64 KB      9.53%
        IDT_LIST:          0 GB         2 KB      0.00%

[130/131] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: cortex-m3
qemu-system-arm: warning: nic stellaris_enet.0 has no peer
*** Booting Zephyr OS build zephyr-v2.5.0-969-g2acec93ba6e7  ***
Running test suite mheap_api
===================================================================
START - test_mheap_malloc_free
 PASS - test_mheap_malloc_free
===================================================================
START - test_mheap_calloc

    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/mem_heap/mheap_api_concept/src/test_mheap_api.c:124: test_mheap_calloc: (mem is NULL)
calloc operation failed
 FAIL - test_mheap_calloc
===================================================================
START - test_mheap_malloc_align4
 PASS - test_mheap_malloc_align4
===================================================================
START - test_k_aligned_alloc
 PASS - test_k_aligned_alloc
===================================================================
START - test_sys_heap_mem_pool_assign

    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/mem_heap/mheap_api_concept/src/test_mheap_api.c:183: test_sys_heap_mem_pool_assign: (ptr is NULL)
bytes allocation failed from system pool
 FAIL - test_sys_heap_mem_pool_assign
===================================================================
START - test_malloc_in_isr

    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/mem_heap/mheap_api_concept/src/test_mheap_api.c:25: tIsr_malloc_and_free: (ptr is NULL)
bytes allocation failed from system pool
 FAIL - test_malloc_in_isr
===================================================================
START - test_malloc_in_thread
 PASS - test_malloc_in_thread
===================================================================
Test suite mheap_api failed.
===================================================================
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Linux leno 5.10.0-3-amd64 #&#x2060;1 SMP Debian 5.10.13-1 (2021-02-06) x86_64 GNU/Linux
 - Toolchain: Zephyr SDK 0.12.3
 - Commit SHA or Version used: 2acec93ba6e70d2de5ad83dbdfb191b7f1424583

andyross, 2021-03-09:
I don't understand.  Some of the tests need to allocate memory from the system pool, CONFIG_HEAP_MEM_POOL_SIZE controls the size of of the system pool.  Apparently you discovered that it needs at least 184 bytes (== 23 chunks in this situation) to pass.

That all sounds expected to me.  Why do you think 64 bytes should be enough, exactly?  Am I not understanding a detail somewhere?

andyross, 2021-03-09:
CONFIG_HEAP_MEM_POOL_SIZE is the number of bytes allocated to a general heap provided by the system as k_malloc() et. al.  The system itself does not use it for the most part.

jharris-intel, 2021-03-09:
>  IOW, should void main(void) {k_malloc(1);} work with CONFIG_HEAP_MEM_POOL_SIZE=64?

There will always be a certain amount of fixed malloc overhead, I don't know how much exactly. It wouldn't surprise me if it was up to a few hundred bytes, especially if the allocator is using a complex scheme under the hood.

chen-png, 2021-03-24:
for this testcase "tests/kernel/mem_heap/mheap_api_concept", the test_mhead_calloc is designed to allocate 8*16 memory size which has been set in source code test_mhead_api.c, why do you think 64 is enough?

chen-png, 2021-03-28:
@yashi the detailed heap memory pool size is related to your testcase and how much memory you want to allocate. 
when you ran this "tests/kernel/mem_heap/mheap_api_concept" with CONFIG_HEAP_MEM_POOL_SIZE=64, it failed at "test_mheap_calloc" due to that this test needs to allocate 8*16=128 bytes, for this memory block, it needs 17 free chunks(8 bytes each chunk) at least(because it needs 4 bytes chunk header), and if sys_heap wants to support 17 free chunks, the bucket_units in heap header chunk should be 5 at least, so the heap header chunk should be 5 chunks at least(sizeof(struct z_heap) + 5 * sizeof(struct z_heap_bucket) = 36 bytes), so the final heap memory size should be at least 180 bytes(header chunk(5*8=40) + free_chunk(17*8=136) + end_chunk(4 bytes)).
for qemu_cortex_m3, the initialized heap memory start address is 0x20000524, it needs to align with 8 bytes, so the final heap memory size should increase 4 bytes, that is to say, this test only will be passed when the CONFIG_HEAP_MEM_POOL_SIZE>=184.

chen-png, 2021-03-28:
and I think this is not a bug in actually, because we need enough heap size if we want to test this testcase, the original CONFIG_HEAP_MEM_POOL_SIZE=256 is ok.

yashi, 2021-03-29:
OK, I've spent some time to read the heap allocator.

My conclusion is that you need at least 28 bytes heap to successfully call `malloc(0)`.

It depends on the arch (or to be more precise, the size of a pointer and location of the memory you pass to `sys_heap_init()`) but, at the best, the size of a heap must be larger than or equal to 28 bytes.  The heap must have 2 chunks for 16 bytes z_heap itself, at least one chunk for user allocation, and the 4 bytes for the footer on 32 bit arch (I'm not sure who's using the footer just yet).

To satisfy this, you need:

 - 3 chunks == 24 bytes
 - a footer == 4 bytes (on 32 bit CPU)
 - thus, total 28 bytes

This assumes that buffer is already aligned to 8 byte alignment.

`sys_heap_init()` has an assertion for the check above but unfortunately it was not printing anything even with `CONFIG_EARLY_CONSOLE=y` and `CONFIG_ASSERT_VERBOSE=y` because `uart_console_init` is at a higher address than `sys_heap_init` on my env. Both are `PRE_KERNEL_1`.

To improve the situation, we can

  a) make the build fail if `K_HEAP_DEFINE()` is given a small size
  b) make `printk()` a special so that we can get more log in PRE_KERNEL_1

Any feedback welcome.

yashi, 2021-03-29:
for `mheap_api_concept`, thank you all for explanation.  I understand that you need some memory to perform the test and it's not a bug.

chen-png, 2021-03-31:
closed this as this is not a bug, also this is duplicate with this issue #33009.

chen-png, 2021-03-31:
> a) make the build fail if `K_HEAP_DEFINE()` is given a small size.
> b) make `printk()` a special so that we can get more log in PRE_KERNEL_1.
> c) document how much overhead we have to pay for each heap.

I am not sure whether printk already worked in zephyr to print those error message with logging system. But indeed, we need to find a way to report those error message when we using a very small heap size.

closed 2021-03-31 by chen-png