zephyrproject-rtos/zephyr #33171: Create Renesas HAL

julien-massot, 2021-03-09:
## Introduction

I wrote a port for Renesas R-Car Gen3 co-processor based on a Cortex-R7
MCU. The plan is to submit these patches to the Zephyr project.

### Problem description

This Renesas platform doesn't use a HAL at the moment, and so
registers offset are defined directly in the Renesas drivers.
Such as https://github.com/zephyrproject-rtos/zephyr/pull/31307/files#diff-d874fb417f530f72748e4e791e5eb881052fc3c38b86737a26b9723645066938R36

### Proposed change

To not have to define registers offset directly in Zephyr drivers, the plan is to
define these offsets and other constants in a specific repository: hal_renesas.

So I'm requesting a https://github.com/zephyrproject-rtos/hal_renesas repository
to store the renesas HAL as it is done for other ARM platforms.

## Detailed RFC

This repository will be an original repository, since there is no existing HAL
for Renesas R-Car SoC. The code will be distributed under Apache-2 license.
It will first target R-Car Gen3 based SoC such as H3,M3,E3 SoCs, and may be
extended to support the futur Gen4 SoC series.

This layer will be register definitions only in a first time.

A first repository has been created here https://github.com/iotbzh/zephyr-hal-renesas
in order to move register definition that have been pull requested here:
https://github.com/zephyrproject-rtos/zephyr/pull/31307

### Dependencies

None

galak, 2021-03-09:
If there isn't a vendor HAL I don't think we should create one.  Instead the defines and such should just live with the drivers as native Zephyr drivers.

julien-massot, 2021-03-09:
I tend to follow the @galak point of view, but as a new comer I'm totally open regarding the subject.

From what I see In Linux tree these #define are pretty stables across the differents SoC series,
and should be for the next one. So it should be manageable without so much #ifdef in the Zephyr code base.
If this argument begin to be wrong, or if there is a mutual effort between the different RTOS which is very plausible
a HAL introduction may have to be reconsidered.

MaureenHelm, 2021-03-09:
> If there isn't a vendor HAL I don't think we should create one. Instead the defines and such should just live with the drivers as native Zephyr drivers.

I agree. We also have a policy that modules are for externally maintained projects, and that code written for Zephyr should be contributed to the main tree.
https://docs.zephyrproject.org/latest/guides/modules.html#modules-external-projects

mnkp, 2021-03-11:
> If there isn't a vendor HAL I don't think we should create one. Instead the defines and such should just live with the drivers as native Zephyr drivers.

I also agree with the above. However, if we expect that there will be differences between Renesas SoC series, it may be worth keeping all the register related defines in header files located somewhere in `soc/arm/renesas_rcar/`. I'm not familiar with the platform but often the same registers have different addresses on different SoC series (we get the base address of the module from DT but offsets within the modules may vary) or some registers are missing. Renesas R-Car family looks large.

julien-massot, 2021-03-11:
Indeed the R-Car family is large. What I can tell is that these registers are valid for D3, E3, M3, H3, anv V3 SoC within the Gen 3 SoC series. Differences may occurs when dealing with clock tree and pinctrl. which will require a file per SoC. About the next Renesas R-Car SoC series I don't have the registers definition yet, but from what I see in Linux for supporting a V3 Gen4 based SoC it shares the same drivers without registers offset modifications. Let see !

For example Pinmux configuration is different per SoC, here is an example on how I plan to manage it for E3 and H3 platforms:
https://github.com/iotbzh/zephyr/blob/renesas-next/soc/arm/renesas_rcar/gen3/pfc-r8a77951.h
https://github.com/iotbzh/zephyr/blob/renesas-next/soc/arm/renesas_rcar/gen3/pfc-r8a77990.h

carlescufi, 2021-03-11:
> I agree. We also have a policy that modules are for externally maintained projects, and that code written for Zephyr should be contributed to the main tree. [...]

+1. If the code is written for Zephyr, then it should be in the main tree. Unless you can find an SDK from Renesas themselves that provides all the register definitions, then we just keep the code in-tree.

carlescufi, 2021-03-11:
Closing this as won't do. Please submit the code directly to the main tree.
Thank you for the contributions!

closed 2021-03-11 by carlescufi