zephyrproject-rtos/zephyr #33173: tests/kernel/workq/work_queue fails on sam_e70_xplained

nashif, 2021-03-09:
**Describe the bug**
Test fails to complete on the sam_e70_xplained board

**To Reproduce**
`./scripts/twister -N --device-testing --hardware-map ~/map.yml -c -p sam_e70_xplained --platform-reports -v -T tests/kernel/workq`

```
*** Booting Zephyr OS build zephyr-v2.5.0-980-g6c3674cd4ac4  ***
Running test suite workqueue
===================================================================
START - test_sequence
- Initializing test items
- Submitting test items
- Submitting work 1 from preempt thread
- Running test item 1
- Submitting work 2 from coop thread
- Submitting work 3 from preempt thread
- Submitting work 4 from coop thread
- Running test item 2
- Submitting work 5 from preempt thread
- Submitting work 6 from coop thread
- Waiting for work to finish
- Running test item 3
- Running test item 4
- Running test item 5
- Running test item 6
PASS - test_sequence
===================================================================
START - test_resubmit
Starting resubmit test
- Submitting work
- Waiting for work to finish
- Resubmitting work
- Resubmitting work
- Resubmitting work
- Resubmitting work
- Resubmitting work
- Checking results
PASS - test_resubmit
===================================================================
START - test_delayed
Starting delayed test
- Initializing delayed test items
- Submitting delayed test items
- Submitting delayed work 1 from preempt thread
- Submitting delayed work 3 from preempt thread
- Submitting delayed work 5 from preempt thread
- Waiting for delayed work to finish
- Submitting delayed work 2 from coop thread
- Submitting delayed work 4 from coop thread
- Submitting delayed work 6 from coop thread
- Running delayed test item 1
- Running delayed test item 2
- Running delayed test item 3
- Running delayed test item 4
- Running delayed test item 5
- Running delayed test item 6
- Checking results
PASS - test_delayed
===================================================================
START - test_delayed_resubmit
Starting delayed resubmit test
- Submitting delayed work
- Waiting for work to finish
- Resubmitting delayed work
- Resubmitting delayed work
- Resubmitting delayed work
- Resubmitting delayed work
- Resubmitting delayed work
- Checking results
PASS - test_delayed_resubmit
===================================================================
START - test_delayed_resubmit_thread
Starting delayed resubmit from coop thread test
- Waiting for work to finish
- Resubmitting delayed work with 1 ms
- Resubmitting delayed work with 1 ms
- Resubmitting delayed work with 1 ms
- Resubmitting delayed work with 1 ms
- - Resubmitting delayed work with 1 ms
- Resubmitting delayed work with 1 ms
- Running delayed test item 1
- Running delayed test item 1
- Checking results
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/workq/work_queue/src/main.c:133: check_results: (num_results not equal to num_te
sts)
*** work items finished: 2 (expected: 1)
FAIL - test_delayed_resubmit_thread
===================================================================
START - test_delayed_cancel
Starting delayed cancel test
- Cancel delayed work from preempt thread
- Waiting for work to finish
- Cancel delayed work from coop thread
- Checking results
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/workq/work_queue/src/main.c:133: check_results: (num_results not equal to num_te
sts)
*** work items finished: 2 (expected: 0)
FAIL - test_delayed_cancel
===================================================================
START - test_delayed_pending
Starting delayed pending test
- Check pending delayed work when in workqueue
- Running delayed test item 1
- Checking results
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/workq/work_queue/src/main.c:133: check_results: (num_results not equal to num_te
sts)
*** work items finished: 3 (expected: 1)
FAIL - test_delayed_pending
===================================================================
START - test_triggered
Starting triggered test
- Initializing triggered test items
- Submitting triggered test items
- Submitting triggered work 1
- Submitting triggered work 2
- Submitting triggered work 3
- Submitting triggered work 4
- Submitting triggered work 5
- Submitting triggered work 6
- Triggering test items execution
- Triggering work 1
- Triggering work 2
- Triggering work 3
- Triggering work 4
- Triggering work 5
- Triggering work 6
- Running triggered test item 1
- Running triggered test item 2
- Running triggered test item 3
- Running triggered test item 4

```

**Expected behavior**
test passes

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows): Linux
 - Toolchain (e.g Zephyr SDK, ...) 0.12.3
 - Commit SHA or Version used: 6c3674cd4ac4fd25fc02b7621a192f37c83a7a14

PerMac, 2021-03-09:
Also fails on nrf platforms

pabigot, 2021-03-09:
Yes, it failed at the time the PR was merged.  Apparently something had changed in the baseline since the last time it was tested.  I'll look into it.

PR #33183 (2021-03-10): tests: kernel: work_queue: remove unpassable tests
The original implementation of resubmitting a delayed work item removed the item not only from the schedule, but also from the work queue if it was already in the work queue.  This is not the semantics of the new implementation, which will leave the work item in the queue if the previous deadline had elapsed and the work item was submitted.

The new semantics is preferred, as it improves consistency with SMP targets where once an item has been submitted to a queue it can be run at any time, and scheduling it again doesn't magically reverse the [... cut]