zephyrproject-rtos/zephyr #33176: tests: kernel: Multiple test cases from tests/kernel/workq/work_queue are failing

PerMac, 2021-03-09:
**Describe the bug**
Multiple test cases from tests/kernel/workq/work_queue are failing on nrf platforms. The tests report the same error: 
```
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/workq/work_queue/src/main.c:133: check_results: (num_results not equal to num_tests)
```
The output also looks weird as if multiple threads were running simultaneously on board with mixed outputs and multiple calls of the same test case and restarting of the board.

**To Reproduce**
Steps to reproduce the behavior:
1. have e.g. nrf52840dk connected to /dev/ttyACM0
2. go to zephyr dir
3. call `./scripts/twister -T /home/maciej/zephyrproject2/zephyr/tests/kernel/workq/work_queue/ -p nrf52840dk_nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs --platform-reports --jobs 1`
4. See error

**Expected behavior**
Test passes

**Impact**
Not clear

**Logs and console output**

<details>
<summary>Click to expand the output</summary>
```
*** Booting Zephyr OS build zephyr-v2.5.0-977-g0b55037d6391  ***
Running test suite workqueue
===================================================================
START - test_sequence
- Initializing test items
- Submitting test items
- Submitting work 1 from preempt thread
- Running test item 1
- Submitting work 2 from coop thread
- Submitting work 3 from preempt thread
- Submitting work 4 from coop thread
- Running test item 2
- Submitting work 5 from preempt thread
- Submitting work 6 from coop thread
- Waiting for work to finish
- Running test item 3
- Running test item 4
- Running test item 5
- Running test item 6
PASS - test_sequence
===================================================================
START - test_resubmit
Starting resubmit test
- Submitting work
- Waiting for work to finish
- Resubmitting work
- Resubmitting work
- Resubmitting work
- Resubmitting work
- Resubmitting work
- Checking results
PASS - test_resubmit
===================================================================
START - test_delayed
Starting delayed test
- Initializing delayed test items
- Submitting delayed test items
- Submitting delayed work 1 from preempt thread
- Submitting delayed work 3 from preempt thread
- Submitting delayed work 5 from preempt thread
- Waiting for delayed work to finish
- Submitting delayed work 2 from coop thread
- Submitting delayed work 4 from coop thread
- Submitting delayed work 6 from coop thread
- Running delayed test item 1
[... 2206 lines cut ...]
PASS - test_resubmit
===================================================================
START - test_delayed
Starting delayed test
- Initializing delayed test items
- Submitting delayed test items
- Submitting delayed work 1 from preempt thread
- Submitting delayed work 3 from preempt thread
- Submitting delayed work 5 from preempt thread
- Waiting for delayed work to finish
- Submitting delayed work 2 from coop thread
- Submitting delayed work 4 from coop thread
- Submitting delayed work 6 from coop thread
- Running delayed test item 1
- Running delayed test item 2
```
</details>

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain Zephyr SDK 0.12.2
 - Commit SHA or Version used zephyr-v2.5.0-977-g0b55037d6391

**Additional context**
Add any other context about the problem here.

nashif, 2021-03-09:
duplicate of #33173

PerMac, 2021-03-09:
I started writing this when the duplicated one was not yet available :) Closing this one

closed 2021-03-09 by PerMac