zephyrproject-rtos/zephyr #33192: LoRaWAN - Application fails to start if module is not powered

ballardr, 2021-03-09:
LoRaWAN - Application fails to start if module is not powered
------

**Describe the bug**
When a LoRa module is attached, LoRaWAN (and other associated switches in proj.conf) are enabled, but power is not enabled to the LoRa module, the main thread fails to run the user code (that inside main() ). It also does not print the banner (*** Booting Zephyr OS build zephyr-v2.4.0-3155-gc19c6fb438b9  ***). Note, if the module is not attached, the system will start normally (even with proj.conf switches saying LoRaWAN is enabled). 

**To Reproduce**
Connect LoRa module to MCU but don't power on boot of MCU.

**Expected behavior**
It should error, not catastrophoic fail. The initilisation should really be done in application code, not via SYS_INIT() as this scenario of power switching external devices (LoRa module in this case) would be very common, especially for devices where very low power draw in standby is desired.

**Impact**
Catastrophoic in terms of a data logger which could have a failed transciever but can still log to memory. Wasted power usage on very low power systems if required to leave module always powered.

**Logs and console output**
No Console output. I enabled:
CONFIG_LOG=y
# 4-DEBUG, 3-INFO, 2-WARNING, 1-ERROR, 0-OFF
CONFIG_LOG_MAX_LEVEL=4
CONFIG_LOG_BACKEND_UART=y
CONFIG_LOG_RUNTIME_FILTERING=y
CONFIG_LOG_BUFFER_SIZE=2048
CONFIG_LOG_PRINTK=y
CONFIG_LOG_PROCESS_TRIGGER_THRESHOLD=0
CONFIG_COVERAGE=n

yet got no logging on the serial interface (I presume there is something I missed).

**Environment (please complete the following information):**
 - Windows 10
 - Zephyr with gnu-arm-embedded
 - zephyr-v2.4.0-3155-gc19c6fb438b9

**Additional context**
In my application, I'm using LoRa on a battery system which we're hoping to last 5-10 years on battery (with battery charging to extend battery life). As such we need to minimise battery usage as much as possible. I'm using an nrf52840dk with a SX1276MB1LAS shield and another board between them. This extra board, one of it's functions is to switch power rails. These rails default to being off, so when the MCU is in deep sleep (system off), the rails are disabled and thus don't waste power. The system is woken up from an external RTC alarm interrupt.

I've tested the behaviour of the bug over multiple reboots changing the power status with a switch. The switch allows 3v3 on boot to the SX1276 module or has the power controlled by the application code (which is the desired behaviour now, and requried for the final design).

Other notes:
The shell boots fine, and I can see the "main" thread is running:

Scheduler: 4450839 since last call
Threads:
 0x20000e10 sysworkq
        options: 0x0, priority: -1 timeout: 536874636
        state: pending
        stack size 1024, unused 904, usage 120 / 1024 (11 %)

*0x20000bd0 shell_uart
        options: 0x0, priority: 14 timeout: 536874060
        state: queued
        stack size 2048, unused 1168, usage 880 / 2048 (42 %)

 0x20000b18 logging
        options: 0x0, priority: 14 timeout: 536873876
        state: pending
        stack size 768, unused 656, usage 112 / 768 (14 %)

 0x20000c88 idle 00
        options: 0x1, priority: 15 timeout: 536874244
        state:
        stack size 320, unused 248, usage 72 / 320 (22 %)

 0x20000d40 main
        options: 0x1, priority: 0 timeout: 536874428
        state: pending
        stack size 1024, unused 760, usage 264 / 1024 (25 %)

LoRa kconfigs enabled are:
CONFIG_SPI=y
CONFIG_SPI_1=y
CONFIG_LORA=y
CONFIG_LORA_SX12XX=y
CONFIG_LORAWAN=y
CONFIG_LORAMAC_REGION_AS923=y

ballardr, 2021-03-09:
@Mani-Sadhasivam, I believe you are the maintainer for the LoRaWAN, LoRa and SX12xx modules

ballardr, 2021-03-09:
I tried enabling the power to the module after the system had been in the not working mode (ie booted without power, and the user code was thus not running), and at that point got a log message:
[00:26:43.065,704] <inf> sx1276: SX1276 Version:00 found
and then the user code started to run.

This may be a similar issue to one I had, where in my case the system would not boot if the kernel could not pull a communication bus into the correct state when trying to send a packet. ie the powered down device, pulled the bus down to ground and thus no messages could be sent.

Mani-Sadhasivam, 2021-03-11:
@ballardr This issue is very generic and not specific to lora. I'm currently away from my dev board, so please test the below patch and see if it fixes your issue:

```
diff --git a/drivers/spi/spi_context.h b/drivers/spi/spi_context.h
index 62805e9367..ef39ed2d64 100644
--- a/drivers/spi/spi_context.h
+++ b/drivers/spi/spi_context.h
@@ -114,11 +114,11 @@ static inline int spi_context_wait_for_completion(struct spi_context *ctx)
        int status = 0;
 #ifdef CONFIG_SPI_ASYNC
        if (!ctx->asynchronous) {
-               k_sem_take(&ctx->sync, K_FOREVER);
+               k_sem_take(&ctx->sync, K_MSEC(10));
                status = ctx->sync_status;
        }
 #else
-       k_sem_take(&ctx->sync, K_FOREVER);
+       k_sem_take(&ctx->sync, K_MSEC(10));
        status = ctx->sync_status;
 #endif /* CONFIG_SPI_ASYNC */
```

ballardr, 2021-03-11:
Thanks @Mani-Sadhasivam, that fixed the issue. When will the patch be incorporated into the master branch? I better not close until it's patched.

Mani-Sadhasivam, 2021-03-11:
> Thanks @Mani-Sadhasivam, that fixed the issue. When will the patch be incorporated into the master branch? I better not close until it's patched.

Thanks for testing! I'll submit a PR and link this issue so that the closing can happen when the PR is merged.

PR #33230 (2021-07-15): drivers: spi: Use timeout for transfer completion
Instead of waiting forever for the SPI transfer to complete, let's use
a timeout value and bail out if elapsed. The timeout value logic is,

xfer_len/frequency + 200 ms (tolerance)

Fixes: #33192
commit fa2d1ea4 (2021-07-15): drivers: spi: Use timeout for transfer completion

Instead of waiting forever for the SPI transfer to complete, let's use
a timeout value and bail out if elapsed. The timeout value logic is,

xfer_len/frequency + tolerance

Tolerance value can be modified using a Kconfig symbol,
CONFIG_SPI_COMPLETION_TIMEOUT_TOLERANCE. It defaults to 200ms.

Fixes: #33192
closed 2021-07-15 by cfriedt