zephyrproject-rtos/zephyr #33196: I2C doesn't work on STM32F103RE

ycsin, 2021-03-10:
**Describe the bug**
I am trying to write a driver for Kionix KX22 I2C accelerometer to be used my STM32F103RET6 custom board. Prior to this I've not tried the I2C function on Zephyr RTOS. Upon testing the KX022 driver, I found out that the I2C doesn't seem to work, it is not even able to generate a START condition.

I have this in my `board_defconfig`:
```
CONFIG_I2C=y
CONFIG_I2C_STM32=y
CONFIG_I2C_STM32_INTERRUPT=y  (more on this later)
```

`board.dts`:
```
&i2c2 {
    pinctrl-0 = <&i2c2_sda_pb11 &i2c2_scl_pb10>;
    clock-frequency = <I2C_BITRATE_FAST>;
    status = "okay";
    accel: kx022@1E {
            compatible = "kionix,kx022";
            label = "KX022";
            reg = <0x1E>;
            status = "okay";
    };
};
```

`prj.conf`:
```
CONFIG_STDOUT_CONSOLE=y
CONFIG_I2C=y
CONFIG_SENSOR=y
CONFIG_KX022=y
# CONFIG_LIS2DH=y
CONFIG_CBPRINTF_FP_SUPPORT=y
CONFIG_LOG=y
CONFIG_I2C_LOG_LEVEL_DBG=y
CONFIG_SENSOR_LOG_LEVEL_DBG=y
```

**To Reproduce**
Steps to reproduce the behavior:
1. Build and flash [lisdh2 sample](https://github.com/zephyrproject-rtos/zephyr/blob/master/samples/sensor/lis2dh/src/main.c) on a board with STM32F103RET6
2. See error

**Expected behavior**
I should be able to see I2C transactions by probing the bus with an oscilloscope. I'm using RIGOL DS1104 which is capable of decoding I2C.

**Impact**
Unable to use I2C related peripheral, unable to proceed with projects.

**Logs and console output**
If I had `CONFIG_I2C_STM32_INTERRUPT` set to 'y'

```
*** Booting Zephyr OS build zephyr-v2.5.0-579-g3b8ea186fd1a  ***
Could not get KX022 device
[00:00:00.502,000] <dbg> i2c_ll_stm32_v1.stm32_i2c_msg_write: stm32_i2c_msg_write: WRITE timeout
[00:00:00.502,000] <dbg> KX022.kx022_init: Failed reading chip id
[00:00:00.502,000] <dbg> KX022.kx022_init: Invalid chip id 0x0
```

If I had `CONFIG_I2C_STM32_INTERRUPT` set to 'n'
```
*** Booting Zephyr OS build zephyr-v2.5.0-579-g3b8ea186fd1a  ***
Could not get KX022 device
[00:00:00.007,000] <dbg> i2c_ll_stm32_v1.msg_end: stm32_i2c_msg_write: ERR 1
[00:00:00.007,000] <dbg> KX022.kx022_init: Failed reading chip id
[00:00:00.007,000] <dbg> KX022.kx022_init: Invalid chip id 0x0
```

The code stuck in this line:
[image]

Peripheral registers:
[image]

**Environment:**
 - OS: Windows 10
 - Toolchain: gcc-arm-none-eabi-10-2020-q4-major-win32
 - Zephyr OS build zephyr-v2.5.0-579-g3b8ea186fd1a

ycsin, 2021-03-10:
I figured out my custom board complicates this thing: The I2C bus is pull up by a GPIO, which was used as a means to turn the accelerometer on or off. I have a `board.c` to initialize the pin to high at `Application` but that was too late as the I2C initializes before that and it has no way of setting the GPIO to high state. I fixed this issue later on by changing the initialization to `POST_KERNEL` and set it to a priority higher than that of I2C initialization and it now works properly: bus pulled high before I2C inits. However, it still doesn't work.

ycsin, 2021-03-10:
In `i2c_ll_stm32.c`

Line 194:

```c
/* Configure dt provided device signals when available */
ret = stm32_dt_pinctrl_configure(cfg->pinctrl_list,
				 cfg->pinctrl_list_size,
				 (uint32_t)cfg->i2c);
```

After the code above the SDA & SCL lines are reset to ground, I'm not sure if this is expected. Then when the clock is enabled in line 209:

```c
if (clock_control_on(clock,
	(clock_control_subsys_t *) &cfg->pclken) != 0) {
	LOG_ERR("i2c: failure enabling clock");
	return -EIO;
}
```

I think the I2C peripheral starts working prematurely(?) and the CR1->SR2's BUSY flag is set.

[image]

According to STM32F1xx ref manual:

>Bit 1 **BUSY**: Bus busy
>- 0: No communication on the bus
>- 1: Communication ongoing on the bus
>
>– Set by hardware on detection of SDA or SCL low
>– cleared by hardware on detection of a Stop condition.
>- It indicates a communication in progress on the bus. This information is still updated when the interface is disabled (PE=0).

However, performing  I2C reset just like line 252 & 253 is also able to clear the **STOP** bit:

```c
LL_I2C_EnableReset(i2c);
LL_I2C_DisableReset(i2c);
```

Therefore it is required to add one more condition to line 244:

```c
#if defined(CONFIG_SOC_STM32F103X8) || defined(CONFIG_SOC_STM32F103XB) || defined(CONFIG_SOC_STM32F103XE)
```

After this the log becomes:

```
*** Booting Zephyr OS build zephyr-v2.5.0-579-g3b8ea186fd1a  ***
Could not get KX022 device
[00:00:00.002,000] <dbg> i2c_ll_stm32_v1.msg_end: stm32_i2c_msg_write: NACK
[00:00:00.002,000] <dbg> KX022.kx022_init: Failed reading chip id
```

I have a reset button connected to the `NRST` pin of the MCU. If I press and release the button swiftly, somehow it is able to initiate the I2C transaction read the WHO_AM_I register, but the axis readings are incorrect, but that could be on me:

```
*** Booting Zephyr OS build zephyr-v2.5.0-579-g3b8ea186fd1a  ***
Polling at 0.5 Hz
#1 @ 13 ms: x -13.825730 , y 2.182254 , z 2.105683
#2 @ 219 ms: x -13.825730 , y 2.182254 , z 2.105683
#3 @ 424 ms: x -13.825730 , y 2.182254 , z 2.105683
#4 @ 630 ms: x -13.825730 , y 2.182254 , z 2.105683
#5 @ 835 ms: x -13.825730 , y 2.182254 , z 2.105683
[00:00:00.003,000] <dbg> KX022.kx022_init: Chip id 0x14
[00:00:00.218,000] <dbg> i2c_ll_stm32_v1.stm32_i2c_generate_start_condition: stm32_i2c_generate_start_condition: START while STOP active!
[00:00:00.423,000] <dbg> i2c_ll_stm32_v1.stm32_i2c_generate_start_condition: stm32_i2c_generate_start_condition: START while STOP active!
[00:00:00.629,000] <dbg> i2c_ll_stm32_v1.stm32_i2c_generate_start_condition#6 @ 1041 ms: x -13.825730 , y 2.182254 , z 2.105683
: stm32_i2c_generate_start_condition: START while STOP active!
#7 @ 1246 ms: x -13.825730 , y 2.182254 , z 2.105683
#8 @ 1452 ms: x -13.825730 , y 2.182254 , z 2.105683
#9 @ 1657 ms: x -13.825730 , y 2.182254 , z 2.105683
#10 @ 1863 ms: x -13.825730 , y 2.182254 , z 2.105683
#11 @ 2068 ms: x -13.825730 , y 2.182254 , z 2.105683
#12 @ 2274 ms: x -13.825730 , y 2.182254 , z 2.105683
#13 @ 2480 ms: x -13.825730 , y 2.182254 , z 2.105683
#14 @ 2685 ms: x -13.825730 , y 2.182254 , z 2.105683
#15 @ 2891 ms: x -13.825730 , y 2.182254 , z 2.105683
#16 @ 3097 ms: x -13.825730 , y 2.182254 , z 2.105683
#17 @ 3302 ms: x -13.825730 , y 2.182254 , z 2.105683
#18 @ 3508 ms: x -13.825730 , y 2.182254 , z 2.105683
#19 @ 3713 ms: x -13.825730 , y 2.182254 , z 2.105683
```

ycsin, 2021-03-10:
I also tried `CONFIG_I2C_STM32_INTERRUPT=y` and the log output is the same as without interrupt.

erwango, 2021-03-10:
@ycsin Thanks for all these details, but I'm still trying to understand what is specific on you're application.

For the record, I2C works on nucleo_f103rb on `samples/shields/x_nucleo_iks01a2` that is performing data read on sevelar I2C sensor devices.

Can you explain what is different in your set up from the above working example ?

ycsin, 2021-03-12:
I just check with a MPU6050 and it works after editing `i2c_ll_stm32.c`'s line 244 to

```c
#if defined(CONFIG_SOC_STM32F103X8) || defined(CONFIG_SOC_STM32F103XB) || defined(CONFIG_SOC_STM32F103XE)
```

So I can confirmed that something is wrong with my driver or `main.c`. Will do a PR soon.

closed 2021-03-12 by ycsin