zephyrproject-rtos/zephyr #33200: USB CDC ACM sample application fails to compile

raveenp, 2021-03-10:
**Describe the bug**
usb cdc sample application using below prj.conf fails to compile,
samples/subsys/usb/cdc_acm/prj.conf

CONFIG_USB_DEVICE_PID is defined with "0x" without proper value, that's creating the compilation failure.

**To Reproduce**
cmake -GNinja -B ./zephyr/samples/subsys/usb/cdc_acm/build/ -DBOARD=bcm958402m2_m7 -DZEPHYR_MODULES=zephyr/cmsis zephyr/samples/subsys/usb/cdc_acm

ninja -C ./zephyr/samples/subsys/usb/cdc_acm/build/

**Expected behavior**
Application should compile without error.

**Impact**

**Logs and console output**

In file included from ../../../../../include/toolchain.h:41,
                 from zephyr/misc/generated/configs.c:7:
zephyr/misc/generated/configs.c: In function '_ConfigAbsSyms':
zephyr/misc/generated/configs.c:11:41: error: invalid suffix "x" on integer constant
   11 | GEN_ABSOLUTE_SYM(CONFIG_USB_DEVICE_PID, 0x);
      |                                         ^~
../../../../../include/toolchain/gcc.h:422:43: note: in definition of macro 'GEN_ABSOLUTE_SYM'
  422 |   "\n\t.type\t" #name ",@object" :  : "n"(value))
      |                                           ^~~~~
[46/119] Building C object zephyr/arch/arch...es/arch__arm__core__aarch64.dir/fatal.c.obj
ninja: build stopped: subcommand failed.

**Environment (please complete the following information):**

**Additional context**

jfischer-no, 2021-03-12:
@raveenp Does you board (bcm958402m2_m7) has USB device driver in Zephyr OS? No?, then why should a USB sample build for it?

jfischer-no, 2021-03-16:
closing it as there is no USB device driver exist for bcm958402m2_m7.

closed 2021-03-16 by jfischer-no