zephyrproject-rtos/zephyr #33208: cbprintf: Package size calculation is using best case alignment

nordic-krch, 2021-03-10:
**Describe the bug**
`cbprintf_package(NULL, 0, ..)` supposes to return required space needed to pack string with arguments. It is using `NULL` address for calculating alignment which is 8 and 16 bytes aligned. However, size depends on initial alignment of the buffer. As a result following code may fail:
```
int len = cbprintf_package(NULL, 0, "test %f", d);
uint8_t *buf = alloca(len); /* buffer not 8 bytes aligned */
len = cbprintf_package(buf, len, "test %f", d);
if (len == -ENOSPC) {
  LOG_ERR("no space");
} 
```

**Expected behavior**
There are few options:
- add requirement that buffer must be aligned to 8 or 16 bytes (it is for long double which is rare, so it may be optional)
- calculate worse case as if buffer was not aligned to 8 or 16 (but require that buffer is 4 bytes aligned)
- when NULL is provided to indicate length calculation use `len` field to indicate alignment of the buffer that will be used.

**Impact**
Unexpected error due to too small buffer.

nordic-krch, 2021-03-10:
@npitre @pabigot what's your view on this?

pabigot, 2021-03-10:
Document that the buffer needs to be aligned and sized based on the requirements of the arguments, and that if not some format operations will fail due to space lost to enforce necessary alignment.

Or stop trying to do alignment in the packed version.  I can see alternating `%c` with `%llu` being a problem (without looking at how that's actually handled).