zephyrproject-rtos/zephyr #33238: tests: drivers: pwm api fails on many stm32 boards

FRASTM, 2021-03-11:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

explora26, 2021-03-11:
The blackpill_f411ce board not work, too.

FRASTM, 2021-03-11:
the test_task() in the test_pwm does not accept the `device_is_ready(pwm_dev)`  but the initial  `if (!pwm_dev)`

erwango, 2021-03-11:
Same issue on frdm_k64k:
```
*** Booting Zephyr OS build zephyr-v2.5.0-1068-gdddfe23b7a8b  ***
Running test suite pwm_basic_test
===================================================================
START - test_pwm_usec
[PWM]: 0, [period]: 2000, [pulse]: 500
E: Unimplemented system call
E: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x00000000
E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x00000000
E:  xpsr:  0x00000000
E: Faulting instruction address (r15/pc): 0x00000000
E: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
E: Current thread: 0x20000060 (unknown)
E: Halting system
```

erwango, 2021-03-11:
@anangl can you have a quick look ?

FRASTM, 2021-03-15:
Thanks @anangl , I confirm this fixes the issue (tested on disco_l475_iot1)

PR #33295 (2021-03-15): kernel: Add missing verification for device_usable_check() system call
so that this function and also device_is_ready() can be called from
user mode.

Fixes #33238.
closed 2021-03-15 by galak