zephyrproject-rtos/zephyr #33253: STM32G4 with USB-C PD: Some pins cannot be used as input by default

martinjaeger, 2021-03-11:
**Describe the bug**

The USB Type-C and Power Delivery Dead Battery feature requires 5k pull-downs on `UCPD1_CC1` and `UCPD1_CC2` pins (`PB4` and `PB6` on STM32G474 series MCUs).

These pull-downs are enabled by default and need to be actively disabled in order to use those pins as inputs.

**To Reproduce**

Configure PB4 or PB6 as inputs using the GPIO API and pull them up externally with e.g. 5k resistor to 3.3V. You will see that they end up in the middle at around 1.65V instead of at 3.3V.

**Expected behavior**

If something is configured as input via the GPIO API it should work out of the box.

These pull-ups seem to be somewhat separate from normal pinmuxing, so I'm not sure where this should be fixed. Possibly in some generic init code for this MCU, as the reference manual says:

> After exiting reset, the USB Type-C “dead battery” behavior is enabled, which may have a pull-down effect on CC1 and CC2 pins. It is recommended to disable it in all cases, either to stop this pull-down or to hand over control to the UCPD1 (which should therefore be initialized before doing the disable)."

**Workaround**

Add following line somewhere in the initialization of the app:

```C
PWR->CR3 |= PWR_CR3_UCPD_DBDIS;
```

str4t0m, 2021-03-16:
I agree that this could be applied in the soc init routine.

For G4 it should indeed be sufficient to set the `PWR->CR3 |= PWR_CR3_UCPD_DBDIS`, as LL includes a function to do exactly we we could use it: 
`LL_PWR_DisableUCPDDeadBattery();`

For G0 Series the same can be accomplished by setting UCPDX_STROBE in SYSCFGR1.
It only affects g071, g0b1, (g081, b0c1). The SOCs g070 and g0b0 don't have a upcd peripheral, but according to the reference manual and to the soc include file have the Strobe bits in the SYSCFG_CFGR1 registers, so it should be set for them as well:
` LL_SYSCFG_DisableDBATT(LL_SYSCFG_UCPD1_STROBE | LL_SYSCFG_UCPD2_STROBE);`

For L5 Series it is the same as for G4.

molchec, 2021-03-30:
> For G4 it should indeed be sufficient to set the `PWR->CR3 |= PWR_CR3_UCPD_DBDIS`, 

it is does not work. pull-down resistor does not turn off for stm32g474ve rev.x

molchec, 2021-03-31:
> @molchec do you confirm that, for G4 :
> 
> * UCPD1 _CC1 and _CC2 pins (PB4 or PB6) are config in ANALOG mode (no pull up/down)
> * UCPD1 clock is enable
> * PWR clock is enabled (__HAL_RCC_PWR_CLK_ENABLE)   to set the PWR_CR3_UCPD_DBDIS bit with  LL_PWR_DisableUCPDDeadBattery()

I solved the problem last night. did not work due to disabled PWR clocking. 
I added 
RCC-> APB1ENR1 | = RCC_APB1ENR1_PWREN;

PR #33970 (2021-04-09): stm32 devices with USB-C PD cannot use CC1 and CC2 pins by default
It disables USB Type-C dead battery pull-down behavior on UCPDx_CC1 and UCPDx_CC2 pins
of the stm32G4 and stm32L5  and stm32G0 soc series which have the USB type C Power Delivery feature. 

After exiting reset, the USB Type-C “dead battery” behavior is enabled, which may have a
pulldown effect on CC1 and CC2 pins. It is recommended to disable it in all cases, either to
stop this pull-down or to hand over control to the UCPD (which should therefore be initialized
before doing the disable).

For stm32l5 and stm32g4, the UCPD DBDI [... cut]
closed 2021-04-09 by nashif