zephyrproject-rtos/zephyr #33261: gatt_notify too slow on Broadcast

CodingGhost, 2021-03-11:
I am using ble_gatt_notify to push data to all connected clients, but it is way to slow if multiple devices are connected.
I am passing NULL as connection pointer.

**To Reproduce**
try to get High throughput (MTU > 260) to multiple clients using bt_gatt_notify

**Expected behavior**
it would be expected, that the speed is halfed with 2 devices, but its more like 10% of what it is with just one device

**Impact**
cannot roll out feature

**Environment (please complete the following information):**
 - Toolchain PlatformIO with ARMGCC
 - V2.5.0

cvinayak, 2021-03-12:
@CodingGhost Please provide details of reproducing the issue like a simple application or modified zephyr sample that I can use for analysis. Tx buffer count are shared across connections, based on how many connections and number of available Tx buffers your throughput will be affected.

CodingGhost, 2021-03-12:
I will try to boil down my app a demo, but It will take some time.

CodingGhost, 2021-03-12:
ok, I kinda sorted it out. a buffer in my code was too small, and because of that the effective packetsize was way too small. Now I have the throughput that I wanted. I dont think it had something todo with Zephyr anymore.

Sorry for the inconvenience caused!

closed 2021-03-12 by CodingGhost