zephyrproject-rtos/zephyr #33265: Power Management Overhaul

nashif, 2021-03-11:
- [ ] Runtime device power management overhaul
   - [ ] Device dependency integration with power management
   - [ ] Address thread concurrency and move away from work queues when calling _get()/_put APIs in device drivers
   - [ ] Address device PM during startup and availability of kernel services
   - [ ] Document on the API level per peripheral the PM related interfaces and when a device needs to be active or when it can be put into a lower power state.
   - [ ] Add samples using hardware for device_pm_get/pull APIs
- [ ] System Level Power Management
  - [ ] Account for wakup times.
  - [ ] Add additional wakeup/sleep time data in DTS per state
  - [ ] Device power management triggered by system/idle thread
  - [ ] Finalize hooks to be implemented by application/SoC.
- [ ] Documentation
- [ ] Testing (Integration with Power analyzers)

nashif, 2021-03-12:
see #14307 for some dependencies..

ceolin, 2021-04-29:
https://github.com/zephyrproject-rtos/zephyr/pull/33942

- Address device PM during startup and availability of kernel services
- Add samples using hardware for device_pm_get/put APIs
- Address thread concurrency and move away from work queues when calling _get()/_put APIs in device drivers (mostly)

ceolin, 2021-05-07:
https://github.com/zephyrproject-rtos/zephyr/pull/34781

nashif, 2021-06-04:
closing this

closed 2021-06-04 by nashif