zephyrproject-rtos/zephyr #33273: The z_smp_reacquire_global_lock() internal API is not used any where inside zephyr code base

enjiamai, 2021-03-12:
**Describe the bug**
The z_prefix internal API, z_smp_reacquire_global_lock() in kernel/smp.c, which is not used by any code in zephyr project.

**To Reproduce**
1. By shell command **grep z_smp_reacquire_global_lock -rn --exclude-dir={doc, twister-out, build}** , you can only find it is declared and defined, but found nowhere it was called.  
2. Run "**twister -T tests/kernel/ -p qemu_x86 --coverage**" to generate the coverage report, and look at the kernel/smp.c line 44~52 were not be executed.

**Expected behavior**
This internal function shall be used by zephyr code base.

**Impact**
The impact is low. But there is no way to be covered by testing unless we call this internal API directly.

**Logs and console output**
Gcov coverage report:
```
44                 :          0 : void z_smp_reacquire_global_lock(struct k_thread *thread)
45                 :            : {
46         [ #  # ]:          0 :         if (thread->base.global_lock_count) {
47                 :          0 :                 arch_irq_lock();
48                 :            : 
49         [ #  # ]:          0 :                 while (!atomic_cas(&global_lock, 0, 1)) {
50                 :            :                 }
51                 :            :         }
52                 :          0 : }
```
**Additional context**
We want to confirm which solution can be applied for this issue:
1. Remove this API (Is it belongs to a dead code?)
2. Design a test case that will call internal API z_smp_reacquire_global_lock(). 
3. Leave it there and use /* LCOV_EXCL_START */   /* LCOV_EXCL_STOP */ to skip it in code coverage report.  

or another better solution for it, thanks!

nashif, 2021-03-12:
I think it is safe to submit a PR and remove this, the usage was remove in 1acd8c2996c3a1ae0691247deff8c32519307f17 if I am not mistaken.

enjiamai, 2021-03-12:
> I think it is safe to submit a PR and remove this, the usage was remove in [1acd8c2](https://github.com/zephyrproject-rtos/zephyr/commit/1acd8c2996c3a1ae0691247deff8c32519307f17) if I am not mistaken.

Got it, thank you! I'll submit a PR for that.

PR #33293 (2021-03-14): kernel: smp: remove unused internal API z_smp_reacquire_global_lock()
The internal function z_smp_reacquire_global_lock() has not used by
anywhere inside zephyr code, the context switch code logic did use it anymore also,  so remove it.

Fixes #33273.
commit 4aed856d (2021-03-14): kernel: smp: Remove unused internal API z_smp_reacquire_global_lock()

The internal function z_smp_reacquire_global_lock() has not used by
anywhere inside zephyr code, so remove it.

Fixes #33273.
closed 2021-03-14 by nashif