zephyrproject-rtos/zephyr #33275: EHL: samples/subsys/shell/shell_module does not work

tbursztyka, 2021-03-12:
samples/subsys/shell/shell_module on EHL does not work at all. 

HW dbg shows it went to the fatal exception handling path.

I disabled all the thread related options in the prj.conf:
```                                                                                            
CONFIG_PRINTK=y
CONFIG_SHELL=y
CONFIG_LOG=y
CONFIG_INIT_STACKS=y
#CONFIG_THREAD_STACK_INFO=y
CONFIG_KERNEL_SHELL=y
#CONFIG_THREAD_MONITOR=y
CONFIG_BOOT_BANNER=n
#CONFIG_THREAD_NAME=y
CONFIG_DEVICE_SHELL=y
CONFIG_POSIX_CLOCK=y
CONFIG_DATE_SHELL=y
#CONFIG_THREAD_RUNTIME_STATS=y
```

And it worked again. So one of this option is broken on EHL.

jhedberg, 2021-03-16:
@tbursztyka I also saw this and worked around it by enabling polling-based input. It seems like a regression since I'm pretty sure this worked when we introduced ehl_crb to the tree. @jenmwms do you know anything about this? How recently have you tested this?

jenmwms, 2021-03-18:
I'm pretty sure it work when we introduced it to the tree, too. I've not personally tested it recently. I'll see what we can find as far as bisecting this.

jenmwms, 2021-03-18:
Just tried this back at fe516b9 - it works (I can use commands like 'help' in the shell module sample).

jenmwms, 2021-03-19:
Ok, bisect and I found it breaks at 7009012 which added `CONFIG_THREAD_RUNTIME_STATS=y` to this samples' `prj.conf`. I'll need to look into configurations depending on this setting to see what may have changed. We can also check the diffs of the .conf before and after.

jhedberg, 2021-03-19:
@jenmwms thanks for bisecting this! I can also confirm that disabling CONFIG_THREAD_RUNTIME_STATS makes things work again. @nashif any idea why that option is not working on EHL?

jenmwms, 2021-03-24:
@nashif this sample works OK on Up2, just tested and can use the sample.

jenmwms, 2021-04-15:
I found that the runtime statistics are gotten using the default kernel timer when using `CONFIG_THREAD_RUNTIME_STATS`. But 
 some architectures, SoCs or boards have timers with higher resolution available via timing functions. Using these timers can be enabled with this instead `CONFIG_THREAD_RUNTIME_STATS_USE_TIMING_FUNCTIONS`. I thought that might be the case for EHL, but had no impact when I changed the config.

jenmwms, 2021-04-29:
@nashif Do you have any ideas why the option `CONFIG_THREAD_RUNTIME_STATS` breaks this shell_module sample for EHL?

nashif, 2021-05-03:
> @jenmwms thanks for bisecting this! I can also confirm that disabling CONFIG_THREAD_RUNTIME_STATS makes things work again. @nashif any idea why that option is not working on EHL?

can you try this:

https://docs.zephyrproject.org/latest/guides/flash_debug/thread-analyzer.html

`west build -b ehl_crb samples/synchronization/ -- -DCONFIG_THREAD_ANALYZER=y -DCONFIG_THREAD_ANALYZER_USE_PRINTK=y -DCONFIG_THREAD_ANALYZER_AUTO=y -DCONFIG_THREAD_ANALYZER_AUTO_INTERVAL=5` 

and see if that works?

jenmwms, 2021-05-04:
Thanks @nashif ! That seems to be working, here's the output:
```
Booting Zephyr OS build zephyr-v2.5.0-3100-g37de0c49878c  ***
thread_a: Hello World from cpu 0 on ehl_crb!
Thread analyze:
 thread_b            : STACK: unused 832 usage 192 / 1024 (18 %); CPU: 0 %
 thread_a            : STACK: unused 392 usage 632 / 1024 (61 %); CPU: 0 %
 thread_analyzer     : STACK: unused 0 usage 512 / 512 (100 %); CPU: 0 %
 idle 01             : STACK: unused 4088 usage 8 / 4096 (0 %); CPU: 0 %
 idle 00             : STACK: unused 4088 usage 8 / 4096 (0 %); CPU: 0 %
Thread analyze:
 thread_b            : STACK: unused 832 usage 192 / 1024 (18 %); CPU: 0 %
 thread_a            : STACK: unused 392 usage 632 / 1024 (61 %); CPU: 0 %
 thread_analyzer     : STACK: unused 0 usage 512 / 512 (100 %); CPU: 0 %
 idle 01             : STACK: unused 4056 usage 40 / 4096 (0 %); CPU: 98 %
 idle 00             : STACK: unused 4088 usage 8 / 4096 (0 %); CPU: 0 %
Thread analyze:
 thread_b            : STACK: unused 832 usage 192 / 1024 (18 %); CPU: 0 %
 thread_a            : STACK: unused 392 usage 632 / 1024 (61 %); CPU: 0 %
 thread_analyzer     : STACK: unused 0 usage 512 / 512 (100 %); CPU: 0 %
 idle 01             : STACK: unused 4056 usage 40 / 4096 (0 %); CPU: 99 %
 idle 00             : STACK: unused 4088 usage 8 / 4096 (0 %); CPU: 0 %

nashif, 2021-05-04:
do we have more details about where it actually fails, i.e. debug the code enabled by  `CONFIG_THREAD_RUNTIME_STATS` in `subsys/shell/modules/kernel_service.c` and see what goes wrong.

jhedberg, 2021-05-05:
```
thread_analyzer     : STACK: unused 0 usage 512 / 512 (100 %); CPU: 0 %
```
Seems like that thread had a stack overflow?

jenmwms, 2021-05-11:
> did you try to increase stacks?

Not yet, thank you. I will try

jenmwms, 2021-05-13:
Noting https://github.com/zephyrproject-rtos/zephyr/pull/35059 helps fix the output from thread analyzer (above comments). Great!

But, no change for this reported failure in samples/.../shell_module with that patch. Same behavior as reported - no output, but removing this lets it work again:
```
- CONFIG_THREAD_RUNTIME_STATS=y

jenmwms, 2021-05-19:
I tried increasing all of the stack sizes and that did not seem to help (used grep in .config to identify all *_STACK_SIZE).

jenmwms, 2021-05-20:
I'm not sure why MUTLIBOOT is configured. Perhaps there is an issue with it being set for OUTPUT_EFI and MULTIBOOT...
CONFIG_MULTIBOOT=y
# CONFIG_MULTIBOOT_INFO is not set
# CONFIG_MULTIBOOT_MEMMAP is not set
# CONFIG_MULTIBOOT_FRAMEBUF is not set
CONFIG_BUILD_OUTPUT_EFI=y

jenmwms, 2021-05-25:
Able to get printk's throughout `arch/x86/core/prep_c.c` using CONFIG_X86_VERY_EARLY_CONSOLE.

jenmwms, 2021-06-02:
@andyross Thanks for the suggestion to try removing this piece from `locore.S`. The sample boots with this removed (and still using the existing THREAD_RUNTIME_STATS=y). 
```
git diff arch/x86/core/intel64/locore.S
diff --git a/arch/x86/core/intel64/locore.S b/arch/x86/core/intel64/locore.S
index 5a56205b59..c7215c452f 100644
--- a/arch/x86/core/intel64/locore.S
+++ b/arch/x86/core/intel64/locore.S
@@ -338,9 +338,7 @@ __resume:
         */
        call z_x86_swap_update_page_tables
 #endif
-#ifdef CONFIG_INSTRUMENT_THREAD_SWITCHING
-       call z_thread_mark_switched_in
-#endif
```
So now to understand why this unblocked us to boot.  Looks like there is a bug here, tho.

jenmwms, 2021-06-11:
So far nothing jumps out as I look in kernel/thread.c related to z_thread_mark_switched_in() or out(). But I do wonder if this is a different bug? I can revisit on Monday:
```
diff --git a/kernel/thread.c b/kernel/thread.c
index b69664f7fe..cfefffd5a1 100644
--- a/kernel/thread.c
+++ b/kernel/thread.c
@@ -1050,7 +1050,7 @@ void z_thread_mark_switched_out(void)
        diff = timing_cycles_get(&thread->rt_stats.last_switched_in, &now);
 #else
        now = k_cycle_get_32();
-       diff = (uint64_t)now - thread->rt_stats.last_switched_in;
+       diff = (uint64_t)(now - thread->rt_stats.last_switched_in);
        thread->rt_stats.last_switched_in = 0;
 #endif /* CONFIG_THREAD_RUNTIME_STATS_USE_TIMING_FUNCTIONS */
```

andyross, 2021-06-14:
Does it still fail if you stub out 100% of the function?  That is, replace it with `void z_thread_mark_switched_in(void) {}`?  It's not impossible this is a stack setup problem that is blowing up due to the simple function call and not anything to do with the content of the function.

If that works, try commenting out individual lines of that function one by one (looking at it, it's an #ifdef hell, so many may not actually be compiled, but whatever).   Eventually you should get to one line that causes the failure.  Then we dig down farther.  (e.g. if it's the SYS_PORT_TRACING_FUNC macro, we know to look in the tracing layer, if it's k_cycle_get_32() then maybe the timer driver isn't initialized yet, etc...)

jenmwms, 2021-06-15:
> So far nothing jumps out as I look in kernel/thread.c related to z_thread_mark_switched_in() or out(). But I do wonder if this is a different bug? I can revisit on Monday:
>
> ```
> diff --git [...]

See https://github.com/zephyrproject-rtos/zephyr/pull/36244.  Note the patch does not resolve this issue in shell module sample.

jenmwms, 2021-06-15:
> Does it still fail if you stub out 100% of the function? That is, replace it with `void z_thread_mark_switched_in(void) {}`? It's not impossible this is a stack setup problem that is blowing up due to the simple function call and not anything to do with the content of the function.

Nice! It boots up and enters sample with the suggestion above. Now diving deeper into that function to find the line and/or case that breaks this. Lots of #ifdefs as mentioned, but making progress.

jenmwms, 2021-06-15:
> k_cycle_get_32() then maybe the timer driver isn't initialized yet, etc...)
After investigating, indeed it seems `k_cycle_get_32()` is causing the issue (failed functional call). So next we look at timer driver, or where/when the timer driver is initialized to maybe see if we can move this thread tracing to after the timer is set up.
https://github.com/zephyrproject-rtos/zephyr/blob/46029914a7ac092f95e586759b8b6cc379af430c/kernel/thread.c#L1018

jenmwms, 2021-06-15:
Ooh! I think I found it. Its related to timing from what I can tell. Easy fix if I'm correct - We need to use `CONFIG_TIMING_FUNCTIONS_NEED_AT_BOOT`, so that we can initialize the timer before we call `switch_to_main_thread`

jenmwms, 2021-06-15:
I was thinking of adding this to the sample prj.conf, but this was only observed on ehl_crb. Is there a better place to specify this config, or no harm to add this for other platforms?

commit b88559a3 (2021-06-16): sample: subsys: shell: shell_module: add config for timing at boot

Use of CONFIG_THREAD_RUNTIME_STATS was added to this sample, but
caused issue because the call to z_thread_mark_switched_in()
is using k_cycle_get_32() before the timer is initialized.
Enable CONFIG_THREAD_RUNTIME_STATS_USE_TIMING_FUNCTIONS=y to
select hidden config CONFIG_TIMING_FUNCTIONS_NEED_AT_BOOT,
which calls to initialize the timer needed at boot for thread
runtime stats configuration.

Fixes #33275
closed 2021-06-16 by jhedberg