zephyrproject-rtos/zephyr #33280: drivers: serial: nrf uarte: The application receives one more byte that was received over UART

tomchy, 2021-03-12:
**Describe the bug**
The application receives one more byte that was received, due to RX counter alignment upon ENDRX event.
In case of the following sequence of UART events:
 - UART_ENDRX
 - UART_RXRDY
 - TIMER_RXTIMEOUT

The application receives one more byte that was received, due to RX counter alignment upon ENDRX event.

**Expected behavior**
The application should receive the exact amount of bytes that was received by the UART peripheral.

**Impact**
This issue may break data transfers, that require at least two UART transmissions, with the delay between them.

**Environment:**
 - OS: Linux
 - Toolchain: NCS SDK
 - NCS v1.3.0

**Additional context**
The issue was found during Zigbee NCP serial transport development. It was mitigated by switching from `UART_ASYNC_API` to the `UART_INTERRUPT_DRIVEN` mode (https://github.com/nrfconnect/sdk-nrf/pull/2835).
After switching, a new way of reproducing the issue was not developed.

XavierChapron, 2021-03-29:
@tomchy @anangl I'm seeing a similar behavior:

I'm using sequences that looks like:
- `mdm_receiver_wake()`
- `mdm_receiver_send()`
- `mdm_receiver_recv()`
- `mdm_receiver_sleep()`

When using UARTE driver, there is an extra char received by the modem_receiver through `mdm_receiver_isr()` 0.5msec after the call to `mdm_receiver_sleep()`.
This extra char always correspond to the last received char.

This doesn't happen when using UART driver.

However I'm not familiar enough with UART and NRF UARTE implementation to give you the exact event sequence.

I've blindly tried your proposed patch #25975 but it didn't work on my use case.

We were able to implement our driver by detecting and ignoring theses extra bytes, but we are still really interested in a real fix!

tomchy, 2021-04-02:
Hi @XavierChapron!
I've rebased the branch and made a draft PR here: https://github.com/zephyrproject-rtos/zephyr/pull/33971

What is your UART driver configuration in the problematic scenario?
Have you tried the timer-based RX byte counting?
```
CONFIG_UART_0_ASYNC=y
CONFIG_UART_0_NRF_HW_ASYNC=y
CONFIG_NRFX_TIMER1=y
CONFIG_UART_0_NRF_HW_ASYNC_TIMER=1
```

When it comes to debugging - I was using the logic probe and toggling GPIO pins on the UART events. Could you grab some traces?
The UART event dispatcher is implemented in the `uarte_nrfx_isr_async` static function. Keep in mind that it is called from an interrupt context, so the pin toggling routine must not use mutexes.