zephyrproject-rtos/zephyr #33291: Using both NET_SOCKETS_SOCKOPT_TLS and POSIX_API fails build

marinbudimir, 2021-03-12:
**Describe the bug**
In my project I am using MQTT TLS so I have `CONFIG_NET_SOCKETS_SOCKOPT_TLS=y`. I also have some Zephyr WiFi drivers which automatically select `POSIX_API`. Build fails while using both of this.

**To Reproduce**
Steps to reproduce the behavior:
1. make any zephyr project
2. put following config in prj.conf file
```
#NET_SOCKETS_SOCOPT_TLS dependecies
CONFIG_NET_SOCKETS=y
CONFIG_NETWORKING=y

CONFIG_NET_SOCKETS_SOCKOPT_TLS=y
CONFIG_POSIX_API=y
```
3. try to build zephyr project
4. see error

**Expected behavior**
Zephyr project builds successfully 

**Logs and console output**
```
[237/293] Building C object modules/mbedtls/CMakeFiles/..__modules__crypto__mbedtls.dir/library/ssl_cli.c.obj
In file included from /home/users/mbudimir/zephyrproject/modules/crypto/mbedtls/include/mbedtls/platform.h:68,
                 from /home/users/mbudimir/zephyrproject/modules/crypto/mbedtls/library/ssl_cli.c:56:
/home/users/mbudimir/zephyrproject/modules/crypto/mbedtls/library/ssl_cli.c: In function 'ssl_generate_random':
/home/users/mbudimir/zephyrproject/modules/crypto/mbedtls/include/mbedtls/platform_time.h:99:24: warning: implicit declaration of function 'time'; did you mean 'gmtime'? [-Wimplicit-function-declaration]
 #define mbedtls_time   time
                        ^~~~
/home/users/mbudimir/zephyrproject/modules/crypto/mbedtls/library/ssl_cli.c:724:9: note: in expansion of macro 'mbedtls_time'
     t = mbedtls_time( NULL );
         ^~~~~~~~~~~~
[239/293] Building C object modules/mbedtls/CMakeFiles/..__modules__crypto__mbedtls.dir/library/ssl_srv.c.obj
In file included from /home/users/mbudimir/zephyrproject/modules/crypto/mbedtls/include/mbedtls/platform.h:68,
                 from /home/users/mbudimir/zephyrproject/modules/crypto/mbedtls/library/ssl_srv.c:56:
/home/users/mbudimir/zephyrproject/modules/crypto/mbedtls/library/ssl_srv.c: In function 'ssl_write_server_hello':
/home/users/mbudimir/zephyrproject/modules/crypto/mbedtls/include/mbedtls/platform_time.h:99:24: warning: implicit declaration of function 'time'; did you mean 'gmtime'? [-Wimplicit-function-declaration]
 #define mbedtls_time   time
                        ^~~~
/home/users/mbudimir/zephyrproject/modules/crypto/mbedtls/library/ssl_srv.c:2479:9: note: in expansion of macro 'mbedtls_time'
     t = mbedtls_time( NULL );
         ^~~~~~~~~~~~
[247/293] Building C object modules/mbedtls/CMakeFiles/..__modules__crypto__mbedtls.dir/library/x509.c.obj
In file included from /home/users/mbudimir/zephyrproject/modules/crypto/mbedtls/include/mbedtls/platform_util.h:61,
                 from /home/users/mbudimir/zephyrproject/modules/crypto/mbedtls/include/mbedtls/cipher.h:64,
                 from /home/users/mbudimir/zephyrproject/modules/crypto/mbedtls/include/mbedtls/oid.h:64,
                 from /home/users/mbudimir/zephyrproject/modules/crypto/mbedtls/library/x509.c:67:
/home/users/mbudimir/zephyrproject/modules/crypto/mbedtls/library/x509.c: In function 'x509_get_current_time':
/home/users/mbudimir/zephyrproject/modules/crypto/mbedtls/include/mbedtls/platform_time.h:99:24: warning: implicit declaration of function 'time'; did you mean 'gmtime'? [-Wimplicit-function-declaration]
 #define mbedtls_time   time
                        ^~~~
/home/users/mbudimir/zephyrproject/modules/crypto/mbedtls/library/x509.c:935:10: note: in expansion of macro 'mbedtls_time'
     tt = mbedtls_time( NULL );
```
```
/home/users/mbudimir/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: modules/mbedtls/lib..__modules__crypto__mbedtls.a(ssl_cli.c.obj):(.literal.mbedtls_ssl_handshake_client_step+0x38): undefined reference to `time'
/home/users/mbudimir/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: modules/mbedtls/lib..__modules__crypto__mbedtls.a(ssl_cli.c.obj): in function `ssl_generate_random':
/home/users/mbudimir/zephyrproject/modules/crypto/mbedtls/library/ssl_cli.c:707: undefined reference to `time'
/home/users/mbudimir/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: modules/mbedtls/lib..__modules__crypto__mbedtls.a(ssl_cli.c.obj): in function `ssl_parse_server_hello':
/home/users/mbudimir/zephyrproject/modules/crypto/mbedtls/library/ssl_cli.c:1894: undefined reference to `time'
/home/users/mbudimir/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: modules/mbedtls/lib..__modules__crypto__mbedtls.a(ssl_srv.c.obj): in function `ssl_parse_client_hello':
/home/users/mbudimir/zephyrproject/modules/crypto/mbedtls/library/ssl_srv.c:2011: undefined reference to `time'
/home/users/mbudimir/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: modules/mbedtls/lib..__modules__crypto__mbedtls.a(ssl_srv.c.obj): in function `ssl_write_server_hello':
/home/users/mbudimir/zephyrproject/modules/crypto/mbedtls/library/ssl_srv.c:2483: undefined reference to `time'
/home/users/mbudimir/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: modules/mbedtls/lib..__modules__crypto__mbedtls.a(x509.c.obj):/home/users/mbudimir/zephyrproject/modules/crypto/mbedtls/library/x509.c:935: more undefined references to `time' follow
```

jukkar, 2021-03-16:
There are conflicts if one selects `CONFIG_POSIX_API` and mbedtls. You can also try if `CONFIG_NET_TCP_ISN_RFC6528=n` if it helps (see commits 51beb28ac9ff53e8aaea1344bdd2abb70a6e75ef and 6526b5b23277a16e5394d506dfab88e29bba8fb4 for some details).

mniestroj, 2021-03-16:
`undefined reference to time` could be solved by #33397.

marinbudimir, 2021-03-17:
> `undefined reference to time` could be solved by #33397.

This solves the problem.

closed 2021-03-17 by marinbudimir