zephyrproject-rtos/zephyr #33303: __ASSERT does not display message or register info in v2.5.0

GavinGonsalves, 2021-03-12:
__ASSERT works fine in v2.4.0, but in v2.5.0 it just hangs.

I'm running the following in a QEMU environment attempting to debug:

    __ASSERT_PRINT("__ASSERT_PRINT \n");
    __ASSERT_LOC(false);
    __ASSERT_MSG_INFO("__ASSERT_MSG_INFO \n");
    __ASSERT_NO_MSG(false);

The failure occurs with __ASSERT_NO_MSG(false)

Example output with v2.5.0:
[10/11] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: qemu64,+x2apic
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.x2apic [bit 21]
SeaBIOS (version zephyr-v1.0.0-0-g31d4e0e-dirty-20200714_234759-fv-az50-zephyr)
Booting from ROM..

Example output with v2.4.0:
__ASSERT_PRINT 
ASSERTION FAIL [0] @ WEST_TOPDIR/apps/src/myFile.c:72
	__ASSERT_MSG_INFO 

ASSERTION FAIL [0] @ WEST_TOPDIR/apps/src/myFile.c:74
[00:00:00.000,000] <err> os: RAX: 0x0000000000000004 RBX: 0x0000000000000000 RCX: 0x0000000000000001 RDX: 0x0000000000000001
[00:00:00.000,000] <err> os: RSI: 0x000000000000004a RDI: 0x000000000010e03d RBP: 0x000000000012a0c0 RSP: 0x000000000012a098
[00:00:00.000,000] <err> os:  R8: 0x0000000000000001  R9: 0x0000000000000039 R10: 0x0000000000000000 R11: 0x0000000000000000
[00:00:00.000,000] <err> os: R12: 0x0000000000000000 R13: 0x0000000000000000 R14: 0x0000000000000000 R15: 0x0000000000000000
[00:00:00.000,000] <err> os: RSP: 0x000000000012a098 RFLAGS: 0x0000000000000246 CS: 0x0018 CR3: 0x0000000000145000
[00:00:00.000,000] <err> os: RIP: 0x0000000000103926
[00:00:00.000,000] <err> os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
[00:00:00.000,000] <err> os: Current thread: 0x0000000000111f80 (main)
[00:00:00.030,000] <err> os: Halting system
FAILED: zephyr/CMakeFiles/run 

If I comment out __ASSERT_NO_MSG(false) in v2.5.0, the following output is seen:
__ASSERT_PRINT 
ASSERTION FAIL [0] @ WEST_TOPDIR/apps/src/myFile.c:72
	__ASSERT_MSG_INFO

galak, 2021-04-06:
@joerchan any updates on this?

joerchan, 2021-04-07:
@GavinGonsalves Some configurations must have changed for you in your application, I'm not able to see the same any issues here.

The dumping of register is not done by the assert handler, but by the fatal exception handler for each architecture. Most of these handlers use the LOG subsystem, so at the very least this must be enabled.
In addition for x86 it seems that `EXCEPTION_DEBUG` needs to be enabled.

I did this change to the hello world application (v2.5.0):
```
--- a/samples/hello_world/src/main.c
+++ b/samples/hello_world/src/main.c
@@ -10,4 +10,12 @@
 void main(void)
 {
        printk("Hello World! %s\n", CONFIG_BOARD);
+
+       __ASSERT_PRINT("__ASSERT_PRINT \n");
+       __ASSERT_LOC(false);
+       __ASSERT_MSG_INFO("__ASSERT_MSG_INFO \n");
+       __ASSERT_NO_MSG(false);
+
+       printk("Hello again!\n");
```
And build and run like this:
```
 west build -b qemu_x86 samples/hello_world -- -DCONFIG_ASSERT=y -DCONFIG_EXCEPTION_DEBUG=y -DCONFIG_LOG=y
 west build -t run
-- west build: running target run
[11/12] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
             RAM:       77856 B         3 MB      2.47%
        IDT_LIST:          0 GB         2 KB      0.00%
[11/12] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: qemu32,+nx,+pae
SeaBIOS (version zephyr-v1.0.0-0-g31d4e0e-dirty-20200714_234759-fv-az50-zephyr)
Booting from ROM..*** Booting Zephyr OS build zephyr-v2.5.0  ***
Hello World! qemu_x86
__ASSERT_PRINT 
ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/samples/hello_world/src/main.c:15
	__ASSERT_MSG_INFO 

ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/samples/hello_world/src/main.c:17
[00:00:00.000,000] <err> os: EAX: 0x00108399, EBX: 0x00000000, ECX: 0x0010d32c, EDX: 0x0010d35c
[00:00:00.000,000] <err> os: ESI: 0x00000000, EDI: 0x00100bd8, EBP: 0x0010d368, ESP: 0x0010d358
[00:00:00.000,000] <err> os: EFLAGS: 0x00000202 CS: 0x0008 CR3: 0x00113000
[00:00:00.000,000] <err> os: call trace:
[00:00:00.000,000] <err> os: EIP: 0x00101837
[00:00:00.000,000] <err> os:      0x00104a1b (0x10d384)
[00:00:00.000,000] <err> os:      0x00100be7 (0x0)
[00:00:00.000,000] <err> os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
[00:00:00.000,000] <err> os: Current thread: 0x10b180 (unknown)
[00:00:00.000,000] <err> os: Halting system
```

GavinGonsalves, 2021-04-07:
@joerchan I can also make it work with the Hello World sample and a minimal proj.conf file of
`CONFIG_ASSERT=y
CONFIG_ASSERT_LEVEL=2`

When I copy the proj.conf file from my application over to Hello World, I get the failure. I'll try to narrow down the CONFIG param that is causing my issue.

GavinGonsalves, 2021-04-07:
@joerchan It is certainly something in the logging config param.

Here is what I currently have that is causing the error:
`CONFIG_LOG=y
CONFIG_LOG_RUNTIME_FILTERING=y
CONFIG_LOG_MODE_OVERFLOW=y
CONFIG_LOG_MAX_LEVEL=4
CONFIG_LOG_DEFAULT_LEVEL=2
CONFIG_LOG_OVERRIDE_LEVEL=2
CONFIG_LOG_PRINTK=y
CONFIG_LOG_PRINTK_MAX_STRING_LENGTH=128
CONFIG_LOG_PROCESS_TRIGGER_THRESHOLD=0
CONFIG_COVERAGE=n
CONFIG_LOG_PROCESS_THREAD=y
CONFIG_LOG_BUFFER_SIZE=65536
CONFIG_LOG_STRDUP_MAX_STRING=128
CONFIG_LOG_STRDUP_BUF_COUNT=64
CONFIG_LOG_BACKEND_FORMAT_TIMESTAMP=y

CONFIG_SOC_LOG_LEVEL_OFF=y
CONFIG_ARCH_LOG_LEVEL_OFF=y
CONFIG_KERNEL_LOG_LEVEL_OFF=y`

If I comment out "CONFIG_KERNEL_LOG_LEVEL", I get the desired result:
`Hello World! qemu_x86
__ASSERT_PRINT 
ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/samples/hello_world/src/main.c:15
	__ASSERT_MSG_INFO 

ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/samples/hello_world/src/main.c:18
	Force ASSERT!
[00:00:00.000,000] <err> os: EAX: 0x001083cc, EBX: 0x00000000, ECX: 0x0011e97c, EDX: 0x0011f0c8
[00:00:00.000,000] <err> os: ESI: 0x00000000, EDI: 0x00100bdc, EBP: 0x0011f0c8, ESP: 0x0011f0b8
[00:00:00.000,000] <err> os: EFLAGS: 0x00000246 CS: 0x0008 CR3: 0x00125000
[00:00:00.000,000] <err> os: call trace:
[00:00:00.000,000] <err> os: EIP: 0x0010183b
[00:00:00.000,000] <err> os:      0x00104fbe (0x11f0e4)
[00:00:00.000,000] <err> os:      0x00100beb (0x0)
[00:00:00.000,000] <err> os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
[00:00:00.000,000] <err> os: Current thread: 0x10b180 (unknown)
[00:00:00.000,000] <err> os: Halting system
FAILED: zephyr/CMakeFiles/run 
`

However, I can get similar passing result by only having the following config:
`CONFIG_ASSERT=y
CONFIG_ASSERT_LEVEL=2

CONFIG_LOG=y
#CONFIG_LOG_RUNTIME_FILTERING=y
#CONFIG_LOG_MODE_OVERFLOW=y
#CONFIG_LOG_MAX_LEVEL=4
#CONFIG_LOG_DEFAULT_LEVEL=2
#CONFIG_LOG_OVERRIDE_LEVEL=2
#CONFIG_LOG_PRINTK=y
#CONFIG_LOG_PRINTK_MAX_STRING_LENGTH=128
#CONFIG_LOG_PROCESS_TRIGGER_THRESHOLD=0
#CONFIG_COVERAGE=n
#CONFIG_LOG_PROCESS_THREAD=y
#CONFIG_LOG_BUFFER_SIZE=65536
#CONFIG_LOG_STRDUP_MAX_STRING=128
#CONFIG_LOG_STRDUP_BUF_COUNT=64
#CONFIG_LOG_BACKEND_FORMAT_TIMESTAMP=y

CONFIG_SOC_LOG_LEVEL_OFF=y
CONFIG_ARCH_LOG_LEVEL_OFF=y
CONFIG_KERNEL_LOG_LEVEL_OFF=y` 

Output:
`[119/120] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: qemu32,+nx,+pae
SeaBIOS (version zephyr-v1.0.0-0-g31d4e0e-dirty-20200714_234759-fv-az50-zephyr)
Booting from ROM..*** Booting Zephyr OS version 2.5.0  ***
Hello World! qemu_x86
__ASSERT_PRINT 
ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/samples/hello_world/src/main.c:15
	__ASSERT_MSG_INFO 

ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/samples/hello_world/src/main.c:18
	Force ASSERT!
FAILED: zephyr/CMakeFiles/run 
`

GavinGonsalves, 2021-04-07:
@joerchan 

Running 2.4.0 with proj.conf:
`CONFIG_ASSERT=y
CONFIG_ASSERT_LEVEL=2

CONFIG_LOG=y
CONFIG_LOG_RUNTIME_FILTERING=y
CONFIG_LOG_MODE_OVERFLOW=y
CONFIG_LOG_MAX_LEVEL=4
CONFIG_LOG_DEFAULT_LEVEL=2
CONFIG_LOG_OVERRIDE_LEVEL=2
CONFIG_LOG_PRINTK=y
CONFIG_LOG_PRINTK_MAX_STRING_LENGTH=128
CONFIG_LOG_PROCESS_TRIGGER_THRESHOLD=0
CONFIG_COVERAGE=n
CONFIG_LOG_PROCESS_THREAD=y
CONFIG_LOG_BUFFER_SIZE=65536
CONFIG_LOG_STRDUP_MAX_STRING=128
CONFIG_LOG_STRDUP_BUF_COUNT=64
CONFIG_LOG_BACKEND_FORMAT_TIMESTAMP=y

CONFIG_SOC_LOG_LEVEL_OFF=y
CONFIG_ARCH_LOG_LEVEL_OFF=y
CONFIG_KERNEL_LOG_LEVEL_OFF=y`

gives the output:
`Booting from ROM..*** Booting Zephyr OS version 2.4.0  ***
Hello World! qemu_x86
__ASSERT_PRINT 
ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/samples/hello_world/src/main.c:15
	__ASSERT_MSG_INFO 

ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/samples/hello_world/src/main.c:18
	Force ASSERT!
[00:00:00.000,000] <err> os: EAX: 0x00108330, EBX: 0x00000000, ECX: 0x0012c2c0, EDX: 0x0012c9cc
[00:00:00.000,000] <err> os: ESI: 0x00000000, EDI: 0x00100f7e, EBP: 0x0012c9cc, ESP: 0x0012c9bc
[00:00:00.000,000] <err> os: EFLAGS: 0x00000246 CS: 0x0008 CR3: 0x00131000
[00:00:00.000,000] <err> os: call trace:
[00:00:00.000,000] <err> os: EIP: 0x001012ce
[00:00:00.000,000] <err> os:      0x00104905 (0x12c9e8)
[00:00:00.000,000] <err> os:      0x00100f8d (0x0)
[00:00:00.000,000] <err> os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
[00:00:00.000,000] <err> os: Current thread: 0x0010a180 (unknown)
[00:00:00.000,000] <err> os: Halting system
FAILED: zephyr/CMakeFiles/run 
`

GavinGonsalves, 2021-04-07:
Here is my Hello World main()
`void main(void)
{
	printk("Hello World! %s\n", CONFIG_BOARD);
    __ASSERT_PRINT("__ASSERT_PRINT \n");
    __ASSERT_LOC(false);
    __ASSERT_MSG_INFO("__ASSERT_MSG_INFO \n");
//    __ASSERT_NO_MSG(false);
    __ASSERT(false, "Force ASSERT!");   // ***** Failure line
    __ASSERT_PRINT("__ASSERT_PRINT 2 \n");
    __ASSERT_LOC(false);
    __ASSERT_MSG_INFO("__ASSERT_MSG_INFO 2 \n");
//__ASSERT_POST_ACTION();
}
`

joerchan, 2021-04-08:
@GavinGonsalves The thing that happened between 2.4.0 and 2.5.0 is that now the KERNEL log level controls the output of the fatal handler.
The change affecting you is this one: https://github.com/zephyrproject-rtos/zephyr/commit/3ed8083dc1625b0f3043b95a3cd6fb1dbd04f84b

So at the very least you need to set the kernel log level to ERR instead of OFF.
`CONFIG_KERNEL_LOG_LEVEL_ERR=y`

PS: You need three ticks to wrap multiline code, like this: ```

joerchan, 2021-04-08:
@nashif Unless we don't wish the kernel log level to disable the fatal error messages then this issue can be closed.

GavinGonsalves, 2021-04-08:
@joerchan I'm good to remove `CONFIG_KERNEL_LOG_LEVEL_OFF`.  Thanks for the help.

However, I would expect an output where I would at least get the "Hello World". Not this:

`-- west build: running target run
[0/1] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: qemu32,+nx,+pae
SeaBIOS (version zephyr-v1.0.0-0-g31d4e0e-dirty-20200714_234759-fv-az50-zephyr)
Booting from ROM..`

There is no real way to figure out what the issue is. This output defeats the purpose of the ASSERT.

This issue is no longer "high" for me. More of a minor improvement.

joerchan, 2021-04-08:
@GavinGonsalves 
From your output here you got it:
```
[119/120] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: qemu32,+nx,+pae
SeaBIOS (version zephyr-v1.0.0-0-g31d4e0e-dirty-20200714_234759-fv-az50-zephyr)
Booting from ROM..*** Booting Zephyr OS version 2.5.0  ***
Hello World! qemu_x86
```

The other stuff is part of QEMU, so it will be different if you use a different board (native_posix for example).

GavinGonsalves, 2021-04-08:
@joerchan That output was with the following config:

`CONFIG_ASSERT=y
CONFIG_ASSERT_LEVEL=2

CONFIG_LOG=y
#CONFIG_LOG_RUNTIME_FILTERING=y
#CONFIG_LOG_MODE_OVERFLOW=y
#CONFIG_LOG_MAX_LEVEL=4
#CONFIG_LOG_DEFAULT_LEVEL=2
#CONFIG_LOG_OVERRIDE_LEVEL=2
#CONFIG_LOG_PRINTK=y
#CONFIG_LOG_PRINTK_MAX_STRING_LENGTH=128
#CONFIG_LOG_PROCESS_TRIGGER_THRESHOLD=0
#CONFIG_COVERAGE=n
#CONFIG_LOG_PROCESS_THREAD=y
#CONFIG_LOG_BUFFER_SIZE=65536
#CONFIG_LOG_STRDUP_MAX_STRING=128
#CONFIG_LOG_STRDUP_BUF_COUNT=64
#CONFIG_LOG_BACKEND_FORMAT_TIMESTAMP=y

CONFIG_SOC_LOG_LEVEL_OFF=y
CONFIG_ARCH_LOG_LEVEL_OFF=y
CONFIG_KERNEL_LOG_LEVEL_OFF=y`

GavinGonsalves, 2021-04-08:
@joerchan Actually, I can't reproduce the output:

-- west build: running target run
[0/1] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: qemu32,+nx,+pae
SeaBIOS (version zephyr-v1.0.0-0-g31d4e0e-dirty-20200714_234759-fv-az50-zephyr)
Booting from ROM..

With the Hello world sample app. I can only see it in my own app. 

I consider this issue resolved.

closed 2021-04-08 by GavinGonsalves