zephyrproject-rtos/zephyr #33322: Questions on ztest : 1) Can twister/ztests run on windows? 2) Project structure

liquiddandruff, 2021-03-14:
## Windows?

I installed the requirements for twister in a virtualenv on Windows but when I try to run it, I get: 

```powershell
PS C:\Users\steven\zephyr> .\venv_windows\Scripts\python.exe C:\Users\steven\.platformio\packages\framework-zephyr\scripts\twister
ZEPHYR_BASE unset, using "C:\Users\steven\.platformio\packages\framework-zephyr"
Traceback (most recent call last):
  File "C:\Users\steven\.platformio\packages\framework-zephyr\scripts\twister", line 209, in <module>
    from twisterlib import HardwareMap, TestSuite, SizeCalculator, CoverageTool, ExecutionCounter
  File "C:\Users\steven\.platformio\packages\framework-zephyr\scripts/pylib/twister\twisterlib.py", line 28, in <module>
    import pty
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.752.0_x64__qbz5n2kfra8p0\lib\pty.py", line 12, in <module>
    import tty
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.752.0_x64__qbz5n2kfra8p0\lib\tty.py", line 5, in <module>
    from termios import *
ModuleNotFoundError: No module named 'termios'
```

So I tried to see how to install `termios` and it seems this is a unix only package. So how does one run ztests on Windows? If it's not possible the docs don't seem to mention this (only tangentially with `platform_allow: native_posix`). 

If that's the case then the docs https://docs.zephyrproject.org/latest/guides/test/ztest.html#quick-start-integration-testing should probably be updated to say ztests cannot run on Windows (and perhaps suggest running them in WSL--which is what I'm trying at the moment).

## Project structure?

Separate question for ztests:

Does twister support running ztests for projects NOT created through `west init`? I.e., out-of-tree projects.

I ask because I understand how the tests in https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/subsys/testsuite/integration are run: each test directory contains `prj.conf`, `testcase.yaml`, `CMakeLists.txt` and the test `src` directory.

These tests can reference the subsystems they intend to test via e.g., `#include <fs>`...

```c
//https://github.com/zephyrproject-rtos/zephyr/blob/master/samples/subsys/fs/littlefs/src/main.c
#include <zephyr.h>
#include <device.h>
#include <fs/fs.h>
#include <fs/littlefs.h>
#include <storage/flash_map.h>
...
```

... and since they all live within `ZEPHYR_BASE`, I presume `twister` does some magic with the `ZEPHYR_BASE` so it knows where the other libraries are defined.

What about in the case of an out-of-tree project, as illustrated in https://docs.zephyrproject.org/latest/application/index.html#creating-an-application ? 

I can't seem to wrap my head around how twister can run ztests in a real project as depicted in `https://docs.zephyrproject.org/latest/guides/test/ztest.html#quick-start-integration-testing` without modifying, say, project CMakeLists.txt. 

For example with the following structure:

```
new_app$ ls
boards/
soc/
CMakeLists.txt
prj.conf
README.rst
includes/
src/
tests/
```

1) What's the expected way to run twister? Add it to path? Use direct paths?
2) Assuming twister is run like below, and the tests in `bar` references sources in `src/`, is this just supposed to work without additional configuration? How?
```
new_app$ twister -T tests/foo/bar/
```

GunZi200, 2021-03-14:
I ran into the same question

> Can twister/ztests run on windows?

The documentation here doesn't make it very clear https://docs.zephyrproject.org/latest/guides/test/twister.html 

For me I tried to follow the instructions and run `source zephyr-env.sh` and ofcourse that didn't work on Windows. Worked well on VMware.

liquiddandruff, 2021-03-14:
Usually .cmd is the windows variant for scripts, but looking at `zephyr-env.cmd` it doesn't seem to do anything.

But now that you mentioned it I see how `twister` is intended to be run now: the env just adds the `ZEPHYR_BASE/scripts` dir to path.

closed 2021-05-20 by nashif