zephyrproject-rtos/zephyr #33324: Using bluetooth hci_usb sample, and set periodic adv enable, but bluez still shows no command supported

super6602, 2021-03-14:
**Describe the bug**
Attempted to use the samples/bluetooth/hci_usb/ project with 
CONFIG_BT_EXT_ADV=y
CONFIG_BT_EXT_ADV_LEGACY_SUPPORT=y
CONFIG_BT_EXT_ADV_MAX_ADV_SET=4
CONFIG_BT_PER_ADV=y
CONFIG_BT_PER_ADV_SYNC=y

to run application over HCI over USB for on a NRF52840-DK. But still shows no periodic adv command supported as below:

>           Read Authenticated Payload Timeout (Octet 32 - Bit 4)
>           Write Authenticated Payload Timeout (Octet 32 - Bit 5)
>           LE Set Data Length (Octet 33 - Bit 6)
>           LE Read Suggested Default Data Length (Octet 33 - Bit 7)
>           LE Write Suggested Default Data Length (Octet 34 - Bit 0)
>           LE Add Device To Resolving List (Octet 34 - Bit 3)
>           LE Remove Device From Resolving List (Octet 34 - Bit 4)
>           LE Clear Resolving List (Octet 34 - Bit 5)
>           LE Read Resolving List Size (Octet 34 - Bit 6)
>           LE Set Address Resolution Enable (Octet 35 - Bit 1)
>           LE Set Resolvable Private Address Timeout (Octet 35 - Bit 2)
>           LE Read Maximum Data Length (Octet 35 - Bit 3)
>           LE Read PHY (Octet 35 - Bit 4)
>           LE Set Default PHY (Octet 35 - Bit 5)
>           LE Set PHY (Octet 35 - Bit 6)
>           LE Enhanced Receiver Test (Octet 35 - Bit 7)
>           LE Enhanced Transmitter Test (Octet 36 - Bit 0)
>           LE Set Advertising Set Random Address (Octet 36 - Bit 1)
>           LE Set Extended Advertising Parameters (Octet 36 - Bit 2)
>           LE Set Extended Advertising Data (Octet 36 - Bit 3)
>           LE Set Extended Scan Response Data (Octet 36 - Bit 4)
>           LE Set Extended Advertising Enable (Octet 36 - Bit 5)
>           LE Read Maximum Advertising Data Length (Octet 36 - Bit 6)
>           LE Read Number of Supported Advertising Sets (Octet 36 - Bit 7)
>           LE Remove Advertising Set (Octet 37 - Bit 0)
>           LE Clear Advertising Sets (Octet 37 - Bit 1)
>           LE Set Extended Scan Parameters (Octet 37 - Bit 5)
>           LE Set Extended Scan Enable (Octet 37 - Bit 6)
>           LE Extended Create Connection (Octet 37 - Bit 7)
>           LE Read Transmit Power (Octet 38 - Bit 7)
>           LE Set Privacy Mode (Octet 39 - Bit 2)
> 

**To Reproduce**
Steps to reproduce the behavior:

1. Build samples/bluetooth/hci_usb/ with CONFIG_BT_EXT_ADV=y, CONFIG_BT_PER_ADV=y and flash to board
2. using bluez "hcitool cmd 0x04 0x02" to get the supported HCI command
3. using bluez "btmon" to see the supported HCI command

**Expected behavior**
bytes37 bit2 ~ bit4, bytes38 bit0 ~ bit2 shall be set to 1

jhedberg, 2021-03-14:
Those all look like Zephyr host options to me. Probably you need to enable some controller options. @cvinayak

cvinayak, 2021-03-14:
https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/controller/hci/hci.c#L618

The bits are not set in the controller, feel free to send a PR.

Yes, you need to enable then under controller Kconfigs, and use in the board specific conf file:

```
CONFIG_BT_CTLR_ADV_EXT=y
CONFIG_BT_CTLR_ADV_PERIODIC=y
CONFIG_BT_CTLR_SYNC_PERIODIC=y
```

Advertising Extensions and dependent features are under development and are for experimental use for now.

jfischer-no, 2021-03-15:
> **Describe the bug**
> Attempted to use the samples/bluetooth/hci_usb/ project with
> CONFIG_BT_EXT_ADV=y
> CONFIG_BT_EXT_ADV_LEGACY_SUPPORT=y
> CONFIG_BT_EXT_ADV_MAX_ADV_SET=4
> CONFIG_BT_PER_ADV=y
> CONFIG_BT_PER_ADV_SYNC=y

This can work with BT_HCI_RAW=y ? Have I missed a significant development?

carlescufi, 2021-03-15:
> This can work with BT_HCI_RAW=y ? Have I missed a significant development?

I don't see why not if the controller is using the exact same ones. But it's a good question, @joerchan?

joerchan, 2021-03-15:
> This can work with BT_HCI_RAW=y ? Have I missed a significant development?

Yes, they should work fine. Setting `CONFIG_BT_EXT_ADV=y` would be wise as that handles some of defaults for buffer sizes `BT_BUF_RX_LEN` for example.

Could ask ourselves if we actually need the BT_CTLR features in this case though. @cvinayak

cvinayak, 2021-03-15:
BT_CTLR features depend on Vendor/SoC support. Let keep h/w depends out of  Host/Common Kconfig. A sample should build clean, and return error codes to indicate if a feature is not supported.

carlescufi, 2021-03-16:
@super6602 could you send a PR updating the necessary bits?

PR #33398 (2021-03-17): bluetooth: controller: Set missing bits in supported list
A bunch of commands were missing from the bitmap in the Read Local
Supported Commands response. Add them accordingly.

Fixes #33324.
commit 66c298db (2021-03-17): bluetooth: controller: Set missing bits in supported list

A bunch of commands were missing from the bitmap in the Read Local
Supported Commands response. Add them accordingly.

Fixes #33324.
closed 2021-03-17 by nashif