zephyrproject-rtos/zephyr #33339: API/functions to get remaining free heap size

jeremyherbert, 2021-03-15:
**Is your feature request related to a problem? Please describe.**
When using heap memory pools, it would be useful during debugging/testing to be able to print out the remaining heap, largest contiguous heap chunk, etc. Basically the same as the Thread Analyzer functionality, but for heap memory.

**Describe the solution you'd like**
Something like:
`uint64_t remaining_bytes = k_heap_remaining(...);`
Or alternatively:
`heap_analyzer_run() // prints some stuff to stdout`

**Describe alternatives you've considered**
I would be open to suggestions as to how I can achieve this with Zephyr in its current state.

npitre, 2021-03-15:
There is `sys_heap_dump()` already, but that presumes familiarity with heap internals.
The first half can be factored out and augmented with actual allocatable sizes.

andyross, 2021-03-16:
FWIW: I think there's an open enhancement request for this already.  FWIW: an option to keep a running count of chunks available in a heap and an API to plumb it out to the user would be really pretty straightforward for anyone who wanted to get their hands dirty.

npitre, 2021-03-16:
Please see PR #33364.

npitre, 2021-03-16:
On Mon, 15 Mar 2021, Andy Ross wrote:

> FWIW: I think there's an open enhancement request for this already.  FWIW: an option to keep a running count of chunks available in a heap and an API to plumb it out to the user would be really pretty straightforward for anyone who wanted to get their hands dirty.

The actual free space doesn't depend 
directly on the number of available chunks though. The largest possible 
allocation might be more useful to applications.

npitre, 2021-03-18:
PR #33364 is merged. I think this pretty much provides what is wanted here.
I'm therefore closing this issue. Please reopen if this isn't satisfactory.

closed 2021-03-18 by npitre

boaks, 2022-08-23:
In my opinion, writing this info just in a dump, doesn't help to include this into a "health" info, which may be even sent over other communication channels then just logging.

boaks, 2022-08-23:
I found `sys_heap_runtime_stats_get`, which seems to be promising for me.

jstuewe23, 2024-04-21:
Yes. That's what I needed as well but it wasn't obvious how to get the system heap pointer.
The example at https://github.com/zephyrproject-rtos/zephyr/issues/33339 worked for me.