zephyrproject-rtos/zephyr #33342: disco_l475_iot1: Multiple definitions of z_timer_cycle_get_32, etc.

pfalcon, 2021-03-15:
```
01:51:29.428 INFO    - 197/398 disco_l475_iot1           tests/subsys/power/power_mgmt/subsys.power.device_pm FAILED Build failure (build)
...
01:52:21.958 /home/buildslave/srv/toolchain/gcc-arm-none-eabi-9-2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: libzephyr.a(cortex_m_systick.c.obj): in function `__NVIC_SetPriority':
01:52:21.958 /home/buildslave/workspace/zephyr-upstream/modules/hal/cmsis/CMSIS/Core/Include/core_cm4.h:1822: multiple definition of `z_clock_driver_init'; libzephyr.a(stm32_lptim_timer.c.obj):/home/buildslave/workspace/zephyr-upstream/zephyr/drivers/timer/stm32_lptim_timer.c:82: first defined here
01:52:21.958 /home/buildslave/srv/toolchain/gcc-arm-none-eabi-9-2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: libzephyr.a(cortex_m_systick.c.obj): in function `z_clock_set_timeout':
01:52:21.958 /home/buildslave/workspace/zephyr-upstream/zephyr/drivers/timer/cortex_m_systick.c:168: multiple definition of `z_clock_set_timeout'; libzephyr.a(stm32_lptim_timer.c.obj):/home/buildslave/workspace/zephyr-upstream/zephyr/drivers/timer/stm32_lptim_timer.c:202: first defined here
01:52:21.958 /home/buildslave/srv/toolchain/gcc-arm-none-eabi-9-2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: libzephyr.a(cortex_m_systick.c.obj): in function `z_clock_elapsed':
01:52:21.958 /home/buildslave/workspace/zephyr-upstream/zephyr/drivers/timer/cortex_m_systick.c:229: multiple definition of `z_clock_elapsed'; libzephyr.a(stm32_lptim_timer.c.obj):/home/buildslave/workspace/zephyr-upstream/zephyr/drivers/timer/stm32_lptim_timer.c:272: first defined here
01:52:21.958 /home/buildslave/srv/toolchain/gcc-arm-none-eabi-9-2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: libzephyr.a(cortex_m_systick.c.obj): in function `z_timer_cycle_get_32':
01:52:21.958 /home/buildslave/workspace/zephyr-upstream/zephyr/drivers/timer/cortex_m_systick.c:242: multiple definition of `z_timer_cycle_get_32'; libzephyr.a(stm32_lptim_timer.c.obj):/home/buildslave/workspace/zephyr-upstream/zephyr/drivers/timer/stm32_lptim_timer.c:300: first defined here
```

Full build log: https://ci.linaro.org/job/zephyr-upstream/6078/PLATFORM=disco_l475_iot1,ZEPHYR_TOOLCHAIN_VARIANT=zephyr,label=docker-xenial-amd64-13/console

FRASTM, 2021-03-15:
this should fix double definition #33285

erwango, 2021-03-15:
@pfalcon thanks for the issue, a PR is submitted, I'm waiting for review...

commit 90d0eb79 (2021-03-16): soc: stm32: Explicitly disable CORTEX_M_SYSTICK if LPTIM enabled

CORTEX_M_SYSTICK should be disabled if LPTIM is selected.
Current implementation is not efficient to do so.
Rework the way the dependency is stated.

Fixes #33342
closed 2021-03-16 by galak