zephyrproject-rtos/zephyr #33351: uart peripheral outputs 7 bits when configured in 8 bits + parity on stm32

nvincent-vossloh, 2021-03-15:
On `nucleo_f746`, when using `UART_CFG_PARITY_EVEN`, the data width is reduced by one bit.
On the following scope capture, I'm sending 0x55 then 0xaa first with parity even, then with parity none. (115200 bauds, 1 stop bit, 8 data bits).
[image]
(serial decoding is configured for 115200 bauds, 1 stop bit, parity even. It is not valid for the two later bytes which have been sent with no parity)

**To Reproduce**
Steps to reproduce the behavior:
1. unzip [test_uart.zip](https://github.com/zephyrproject-rtos/zephyr/files/6109122/test_uart.zip) under zephyr/samples/ : 
2. `west build -p auto -b nucleo_f746zg -d build/nucleo_f746zg/test_uart zephyr/samples/test_uart/`
3. probe on scope `usart6_tx` (pg14/arduino D1)

**Expected behavior**
The frames lengths of the first burst should be one bit longer than the second burst, when configuring `data_bits` to `UART_CFG_DATA_BITS_8`, I'm expecting to see 8 data bits of payload (excluding parity bit), and not 7 bits + parity (=8 bits).

**Environment (please complete the following information):**
- zephyr v2.5.0

I have the feeling that the issue comes from the fact that for (all ? some ?) stm32 devices, the *usart frame formats* table in reference manuals (f746, h743, f091) indicates that the `M` bits will configure the length of the payload which also depends on wether parity is enabled.

I made some modifications in `drivers/serial/uart_stm32.c` available [here](https://github.com/nvincent-vossloh/zephyr/commit/830997ddeded147828aa94c6f8250eaa88b7a423) which help with the situation described above by configuring the M bits based on parity being enabled or not.

If the behaviour described is indeed an issue and not a misunderstanding of the API on my part, I'll submit a PR.

Thanks.

FRASTM, 2021-03-16:
I see that "The frame format consists of a set of data bits in addition to bits for synchronization and optionally a parity bit for error checking..
.The data length can be 9, 8, or 7 bits with the **parity bit counted**. "
[image]

nvincent-vossloh, 2021-03-16:
On STM32 hardware (at least the few ones I checked), yes, the parity bit is counted in the data length, I do not discuss that point.
My question is on zephyr API side, does the `UART_CFG_DATA_BITS_x` macros specify the length with the parity bit counted.

erwango, 2021-03-17:
Assigned to @dcpleung to answer the uart api question

dcpleung, 2021-03-17:
The whole set of bits with data and parity bits is usually referred to as a "data frame". Our API defines data bits for configuration so it should only count the data itself and not other. So `UART_CFG_DATA_BITS_7` only refers to the 7-bit data being transmitted. Parity bit is extra.

closed 2021-04-01 by galak