zephyrproject-rtos/zephyr #33352: Arduino Nano 33 BLE sense constantly resetting.

bkalthoff, 2021-03-15:
**Describe the bug**
After flashing the hello_world sample the board constantly resets over and over.
I've tried using different the master branch of zephyr and the v2.5 branch but to no avail.
I also tried different version of bossac, one that I built from source and one that came with the arduino IDE.
Flashing with the arduino IDE does not cause constant resetting. I have yet to try on building and flashing on another machine.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p -b arduino_nano_33_ble samples/hello_world
2. west flash --bossac="/path/to/bossac"
3. sudo dmesg
4. Check if USB constantly disconnecting and reconnecting.

**dmesg output**

```[ 9152.163254] cdc_acm 1-10:1.0: ttyACM0: USB ACM device
[ 9152.206329] usb 1-10: USB disconnect, device number 112
[ 9152.530787] usb 1-10: new full-speed USB device number 113 using xhci_hcd
[ 9152.689489] usb 1-10: New USB device found, idVendor=2341, idProduct=005a, bcdDevice= 0.11
[ 9152.689490] usb 1-10: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 9152.689491] usb 1-10: Product: Arduino Nano 33 BLE
[ 9152.689492] usb 1-10: Manufacturer: Arduino
[ 9152.689492] usb 1-10: SerialNumber: 0000000000000000A5048AA7799337A8
[ 9152.690955] cdc_acm 1-10:1.0: ttyACM0: USB ACM device
[ 9152.725168] usb 1-10: USB disconnect, device number 113
[ 9153.054786] usb 1-10: new full-speed USB device number 114 using xhci_hcd
[ 9153.213407] usb 1-10: New USB device found, idVendor=2341, idProduct=005a, bcdDevice= 0.11
[ 9153.213408] usb 1-10: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 9153.213409] usb 1-10: Product: Arduino Nano 33 BLE
[ 9153.213410] usb 1-10: Manufacturer: Arduino
[ 9153.213410] usb 1-10: SerialNumber: 0000000000000000A5048AA7799337A8
[ 9153.214946] cdc_acm 1-10:1.0: ttyACM0: USB ACM device
[ 9153.243920] usb 1-10: USB disconnect, device number 114
[ 9153.566790] usb 1-10: new full-speed USB device number 115 using xhci_hcd
[ 9153.725412] usb 1-10: New USB device found, idVendor=2341, idProduct=005a, bcdDevice= 0.11
[ 9153.725413] usb 1-10: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 9153.725414] usb 1-10: Product: Arduino Nano 33 BLE
[ 9153.725414] usb 1-10: Manufacturer: Arduino
[ 9153.725415] usb 1-10: SerialNumber: 0000000000000000A5048AA7799337A8
[ 9153.727355] cdc_acm 1-10:1.0: ttyACM0: USB ACM device
[ 9153.762748] usb 1-10: USB disconnect, device number 115```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain (Zephyr SDK)
 - Board https://store.arduino.cc/arduino-nano-33-ble-sense
 - V2.5 branch and Master branch

bkalthoff, 2021-03-15:
Found success when using UART and finally got some output. Guessing that the cdc_acm driver isn't working correctly.

galak, 2021-03-16:
> Found success when using UART and finally got some output. Guessing that the cdc_acm driver isn't working correctly.

When you say using UART, what do you mean?

@saltyJeff  can you look at this bug?

bkalthoff, 2021-03-16:
> > Found success when using UART and finally got some output. Guessing that the cdc_acm driver isn't working correctly.
> 
> When you say using UART, what do you mean?
> 
> @saltyJeff can you look at this bug?

I managed to get serial output through the tx/rx pins, but not the usb. I tried on two different machines, same issue.

carlescufi, 2021-03-23:
@jfischer-no could you try to shed some light here?

jfischer-no, 2021-03-23:
I do not think this is related to USB device support as it works fine on different nRF52840 based boards.
I also do think it is not a bug but configuration issue. Please check `samples/subsys/shell/shell_module ` build with overlay-usb.conf and `samples/subsys/usb/console`.

jfischer-no, 2021-03-24:
Is it related to #33523?

Stefan-Schmidt, 2021-05-05:
For me it was very much related. I have seen the same resetting behavior and the root was indeed the double offset problem described in #33523

Working around this double offset did lead to successful flashing and no more USB resetting.

carlescufi, 2021-05-06:
Closing as duplicate of #33523

closed 2021-05-06 by carlescufi