zephyrproject-rtos/zephyr #33371: mec15xxevb_assy6853: tests/drivers/gpio/gpio_basic_api/ failed

s-kelvin, 2021-03-16:
**Describe the bug**
drivers.gpio.2pin.gpio_port in tests/drivers/gpio/gpio_basic_api/ failed. 

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b mec15xxevb_assy6853 -p auto tests/drivers/gpio/gpio_basic_api/
2. west flash
3. see error

**Expected behavior**
Test suite should pass.

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v2.5.0-1160-gb8b8e8cd6b1a  ***
Running test suite gpio_basic_test
===================================================================
START - test_gpio_port
Validate device GPIO140_176
Check GPIO140_176 output 15 connected to input 14
FATAL output pin not wired to input pin? (out high => in low)
```

**Environment (please complete the following information):**
 - Linux Ubuntu 18.04.4 LTS
 - zephyr-v2.5.0-1160-gb8b8e8cd6b1a
 - Toolchain zephyr-sdk-0.12.3

scottwcpg, 2021-03-17:
Assume jumper placed on EVB JP41 1-3 to connect GPIO 0156 and 157.
Root cause is a bug in the GPIO driver. Test asks driver to disconnect GPIO 0157. Test then asks driver to set GPIO 0157 to output high. Driver does not clear GPIO control register power gate field leaving the pin powered off (disconnected).

scottwcpg, 2021-03-17:
I submitted a fix in PR#33445

closed 2021-03-18 by nashif