zephyrproject-rtos/zephyr #33378: Extended advertising switch on / switch off loop impossible

ppluta, 2021-03-16:
**Describe the bug**
A loop of bt_le_ext_adv_start() - bt_le_ext_adv_stop() sequences executes properly only once. All subsequent iterations give this warning and error:
[00:00:23.551,635] <wrn> bt_hci_core: opcode 0x2039 status 0x0c
[00:00:23.551,635] <err> bt_hci_core: Failed to start advertiser

followed after few times by:

[00:00:28.552,062] <wrn> bt_hci_core: opcode 0x2039 status 0x07
[00:00:28.552,062] <err> bt_hci_core: Failed to start advertiser

Returns of bt_le_ext_adv_start() are -5 in this case. In the first iteration bt_le_ext_adv_start() returns 0. bt_le_ext_adv_stop() return 0 all the time.

**To Reproduce**
Initialize with:
advertisement_callback contains all 3 available callbacks and all of them are empty functions, only with return.
packet contains one long long int amount of data.
bt_le_ext_adv_create(BT_LE_EXT_ADV_NCONN_NAME, &advertisement_callback, &advertising_set);
bt_le_ext_adv_set_data(advertising_set, packet, ARRAY_SIZE(packet), NULL, 0);

Loop is following:
for(;;)
{
    struct bt_le_ext_adv_start_param parameters = {0, 1};
    bt_le_ext_adv_start(advertising_set, &parameters);
    k_sleep(K_MSEC(100));
    bt_le_ext_adv_stop(advertising_set); 
    k_sleep(K_MSEC(4900));
}

**Expected behavior**
It should simply work every time.

**Impact**
Blocker

**Logs and console output**
00:00:23.551,635] <wrn> bt_hci_core: opcode 0x2039 status 0x0c
[00:00:23.551,635] <err> bt_hci_core: Failed to start advertiser

followed after few times by:

[00:00:28.552,062] <wrn> bt_hci_core: opcode 0x2039 status 0x07
[00:00:28.552,062] <err> bt_hci_core: Failed to start advertiser

**Environment (please complete the following information):**
Latest Zephyr (2.5.99), board Particle Xenon on nRF52840

**Additional context**

cvinayak, 2021-03-19:
@ppluta Please try the latest master. I have tried to use periodic_adv with your said changes and I can not reproduce the issue in babblesim simulation. I can try it on nRF52840DK, but please provide a draft PR, so that I can use your application to know the full application code, base commit, and be able to build at my end on target.

ppluta, 2021-03-24:
Indeed, I confirm. It works with the latest master now.

closed 2021-03-24 by ppluta